osqzss/gps-sdr-sim #246: About accuracy of determined coordinates

hash-hash, 2020-05-21:
Hello.
I have a question about accuracy of determined coordinates by reciever, if it recieves a signal from gps-sdr-sim. By accuracy, I mean the deviation from coordinates which determined during the generation of the signal. I can't check it myself with a real reciever and transmitter now, but i have my matlab model of a gnss reciever. I approved the model by recordings of a real signal almost a year ago and there was not a big error (not bigger than 5-10 meters). But now when I generate a signal and send it to my model, an error of about 30-40 meters appears. I can not once again approve my model right now, but I need to find out why, because it is very important for my undergraduate work.

IvanKor, 2020-05-22:
gps-sdr-sim + hackrf + NEO-M8T
[image]

[Accuracy-GNSS](https://github.com/IvanKor/Accuracy-GNSS)
quadrifilar backfire helix antenna + NEO-M8T
[image]

hash-hash, 2020-05-22:
Thanks for the answer. I saw that you used the 2.619 MHz sample rate in the experiment, right?
I am trying to use a GNSS-SDR receiver with a sampling frequency of 12.276 MHz and only now the error is ~ 0.1-1 m, but the height error is almost 10 meters. In the anti-drone window, I saw that the height of the receiver is 4643.5, and in the signal 4657.8, right?

IvanKor, 2020-05-23:
I used 2.619 MHz in the old version,
latest version has several fixes :
1. residual error from
[image]
to
[image]

2. sample rate UP  to 12 MHz
3. e.t.s.

height  4643.5 --> 4657.8 it is
https://d3i71xaburhd42.cloudfront.net/334560ae5bfd9a538b65cfc8bfcc2a8e45518003/6-Figure5-1.png

closed 2020-06-05 by hash-hash