osqzss/gps-sdr-sim #250: how to generate a bin file for negative longitude values

vipulpjain, 2020-06-18:
Hi,

I am trying to use the below command to generate a bin file (to replay using USRP-B200)

./gps-sdr-sim -e brdc3540.14n -l 38.903057 -77.028091,100 -s 2500000 -b 16 -v

When I use the above command, it gives invalid option error as below:
./gps-sdr-sim: invalid option -- '7'
Usage: gps-sdr-sim [options]

However, if I use the other values for latitude/longitude (without -), it works as expected. 

Any guidance will be appreciated.  

Thanks,
Vipul

osqzss, 2020-06-18:
They have to be separated by commas.

vipulpjain, 2020-06-18:
> They have to be separated by commas.

I tried it as per your suggestion, but still same error. Below command and error. 

./gps-sdr-sim -e brdc3540.14n -l 38.903057, -77.028091,100 -s 2500000 -b 16 -v
./gps-sdr-sim: invalid option -- '7'
Usage: gps-sdr-sim [options]

Thanks,
Vipul

vipulpjain, 2020-06-18:
> > They have to be separated by commas.
> 
> I tried it as per your suggestion, but still same error. Below command and error.
> 
> ./gps-sdr-sim -e brdc3540.14n -l 38.903057, -77.028091,100 -s 2500000 -b 16 -v
> ./gps-sdr-sim: invalid option -- '7'
> Usage: gps-sdr-sim [options]
> 
> Thanks,
> Vipul

Got it. I added a space between , and - .

Thanks for your suggestion.

closed 2020-06-18 by vipulpjain