osqzss/gps-sdr-sim #255: Unstable satellites and traces and DJI Phantom 3 spoofing

alberlv, 2020-07-15:
Good afternoon. First of all, thank you very much for your work and your attention.

I am trying to spoof a DJI Phantom 3 dron **outdoors** and I have achieved it but with some troubles. I am using gps-sdr-sim to generate the binary file, and then I am transmitting it with an Ettus X310 connected to an antenna.

The thing is I can't understand when and why it works. I mean, I am generating and transmitting samples every day, and sometimes it works and sometimes it doesn't work. The main problems I have found are that the dron takes a long time to lock to my fake satellites, but it is very fast to connect to the real ones. Also, when the dron locks to my satellites, the trace is very unstable as you can see in the attached picture. So, the basic questions are:

- Why is my trace so unstable? What am I doing wrong?
- As I am doing all the experiments outside, could the problem be in the coexistence of fake and true satellites? 

Thank you very much in advance. 

[image]

RannyTheCoder, 2020-07-23:
Yes, I am also facing the same issue

jamesl-dm, 2020-11-09:
When the signals appear and disappear, it usually means there is a frequency offset. Your SDR's local oscillator has a discrete step size (ie. ~2.38419 Hz on the Pluto), and the baseband rate has a discrete step size. You need the introduced offsets to be very close to zero, or the receiver will get confused. So you must choose a baseband rate that can be exactly achieved, and output the signal at a frequency that is very close to an achievable frequency.

Some SDR's do automatic PPM compensation, which can be problematic. For instance the Pluto has a xo_corrections variable which is set at the factory, and tweaks the baseband rate and the output frequency to try to compensate for the crystal PPM. But if this is allowed, and the compensation isn't a just the right spot, it can introduce frequency offsets. The GPS receiver knows how to deal with PPM, but it doesn't like frequency offsets that aren't well explained by PPM. So turn the xo_correction off.

atchyuth-rao, 2021-04-21:
i am also facing the same issue, when i tested on DJI mavic pro drone for first time using Pluto SDR for GPS signal generation,

I have observed aircraft traces in remote control as per my user motion file, after that till now i have tested so many times but unable to get the traces

any suggestions to solve this problem, thanks in advance

Shangu-xsg, 2021-04-29:
@jamesl-dm，I'm facing a similar issue and I think your answer is very reasonable, but I don't know how to turn off XO_correction, can you tell me how to turn off XO_correction? I am using GnuRadio to transmit GPS spoofing signals. Looking forward to your reply.

Mictronics, 2021-04-29:
@Shangu-xsg  See https://www.mictronics.de/posts/ADLAM-Pluto-OCXO-Mod/

Shangu-xsg, 2021-04-30:
@Mictronics，hi， thank you for your answer, which successfully helped me solve the problem of how to turn off xo_correction. But after I turned off xo_correction, the problem of not being able to fix remains. I used a modified Pluto and ANT500. When I sent the GPS spoofing signal, the GPS receiver could search for many satellites, but the GPS receiver never used the satellites found. Could you give me some advice on how to solve this problem based on your successful experience? Thank you in advance.
[image]
[image]
[image]
[image]

jamesl-dm, 2021-04-30:
You could try my .grc file, using 4 MHz baseband signals: https://github.com/osqzss/gps-sdr-sim/files/6105847/gps-sdr-sim.zip

Shangu-xsg, 2021-04-30:
@jamesl-dm,hi,I used your.grc file and turned the xo_correction off, but the same result. Are you using an antenna to transmit a signal?

jamesl-dm, 2021-04-30:
Yes - maybe your device has anti-spoofing features?

jamesl-dm, 2021-04-30:
How long is your recording? Needs to be 3+ minutes long, and I recommend 4 MHz bandwidth on the Pluto.

Shangu-xsg, 2021-04-30:
@jamesl-dm ,my device has not anti-spoofing features, and I have occasionally managed to trick my gps receiver into briefly using the satellite I received. 
I used 4mhz bandwidth and recorded it for a long time, but it didn't work. I am now wondering if I succeeded in turning off  xo_correction. Is it possible to turn off the xo_correction in the Plutosdr interface by typing  fw_setenv xo_correction 0?

jamesl-dm, 2021-04-30:
No, you want:
fw_setenv xo_correction 40000000
pluto_reboot reset

Shangu-xsg, 2021-04-30:
Ok,I'll give it a try later. Thank you very much for your help.