osqzss/gps-sdr-sim #261: NASA discontinuing FTP server for ephemeris data.

cmconrad, 2020-09-21:
The README.md suggests downloading the daily GPS ephemeris data from NASA's FTP server. This service is ending on October 31, 2020 as per [their website.](https://cddis.nasa.gov/Data_and_Derived_Products/GNSS/broadcast_ephemeris_data.html) Please update the README to go suggest visiting the new HTTPS instead.

cmconrad, 2020-09-21:
The new way to get ephemeris data is as follows:

1. Register an account with https://urs.earthdata.nasa.gov/ (There is no way around this step).
2. Create a file called `.netrc`  with the following [format](https://cddis.nasa.gov/Data_and_Derived_Products/CreateNetrcFile.html).
`machine urs.earthdata.nasa.gov login <username> password <password>` where \<username> and \<password> are the values you set when you created your Earthdata login account. Don't forget to `sudo chmod 004 .netrc` so that no one can read your file and get your username and password.

3. You can automate with the process with following script.
```
#!/bin/sh
day=$(date +%j)
year=$(date +%Y)
yr=$(date +%y)

curl -c /tmp/cookie -n -L -o "brdc""$day""0.$yr""n.Z" "https://cddis.gsfc.nasa.gov/archive/gnss/data/daily/$year""/brdc/brdc""$day""0.$yr""n.Z"

uncompress "brdc""$day""0.$yr""n.Z"
echo "brdc""$day""0.$yr""n.Z"

```

The `.netrc` file along with the `-n` flag allows curl to automatically authenticate when attempting to retrieve the file.

I hope this helps and please consider expanding the README to include more information like this.

ThatcherC, 2021-03-04:
Thanks @cmconrad - this was super helpful. I've got a couple updates to your script to reflect 1) a different NASA URL and 2) a change in the Ubuntu's OpenSSL settings that gave me trouble running this script and 3) a couple readability changes.

Here's how I get ephemeris now on Ubuntu 20.04: (first two steps are the same as @cmconrad's)

1. Register an account with https://urs.earthdata.nasa.gov/ (There is no way around this step).

2. Create a file called .netrc with the following format:
`machine urs.earthdata.nasa.gov login <username> password <password>` where <username> and <password> are the values you set when you created your Earthdata login account. Don't forget to sudo chmod 004 .netrc so that no one can read your file and get your username and password.

3. Put the following in `~/.openssl_allow_tls1.0.cnf`:
```
openssl_conf = openssl_init

[openssl_init]
ssl_conf = ssl_sect

[ssl_sect]
system_default = system_default_sect

[system_default_sect]
CipherString = DEFAULT@SECLEVEL=1
```
  This is an OpenSSL configuration file that allows you to *temporarily* make cURL use an older, weaker set of SSL settings. In the script below, the `OPENSSL_CONF=~/.openssl_allow_tls1.0.cnf` preceding the `curl` command applies these OpenSSL configs to just that single curl command, so the rest of your system still uses the newer preferred OpenSSL settings. I found this solution in an [AskUbuntu forum answer](https://askubuntu.com/questions/1250787/when-i-try-to-curl-a-website-i-get-ssl-error/1250807#1250807) and it's working well for me on Ubuntu 20.04.

4. Put the following into a script (I called mine `pull-brdc.sh`):
```
#!/bin/sh
day=$(date +%j)
year=$(date +%Y)
yr=$(date +%y)

OUTFILE="brdc""$day""0.$yr""n.gz"
URL="https://cddis.nasa.gov/archive/gnss/data/daily/$year/brdc/$OUTFILE"
echo "Downloading ${URL} into ${OUTFILE}"

OPENSSL_CONF=~/.openssl_allow_tls1.0.cnf curl -c /tmp/cookie --netrc-file ~/.netrc -L -o $OUTFILE $URL

uncompress $OUTFILE
echo $OUTFILE
```
5. Run `chmod +x pull-brdc.sh` and run with `./pull-brdc.sh` and you should get today's brdc file. 

One change I made to the cURL command was to replace the original `-n` flag with ` --netrc-file ~/.netrc`. The behavior is exactly the same, since `~/.netrc` is the default file reading in `-n` mode. However, with the `--netrc-file` flag, you could change the path to the `.netrc` file in case you are on a shared machine and don't want to put your username and password in the home directory.

closed 2021-03-04 by cmconrad