osqzss/gps-sdr-sim #263: Phone is not scanning any satalites

hosam37r, 2020-09-27:
Hi All
I create the *.bin file with running: 
`./gps-sdr-sim -b 8 -e brdc2710.20n -l 32.0244770,35.8971050,100 -s 1250000 -d 100`

Then I transfer the file using:
`hackrf_transfer -t gpssim.bin -f 1575420000 -s 2600000 -x 0`

I put the antenna and a galaxy s6 phone in a metal drawer to make sure it wont get any signals from outside. Also Wi-Fi and Bluetooth were off.
I could not scan any satellites. 
The TX antenna on my HACKRF is solid red. during transfer. 
I read the threads. I tired with -C 10 and -C 4 but no luck

Can someone tell me what am i doing wrong?

osqzss, 2020-09-27:
Use the same sampling rate for both the signal generation and transmission.

hosam37r, 2020-09-28:
Hi 
Thanks for the reply. 
I tried sampling both at 2600000  but still have the same problem.
Thanks,

hosam37r, 2020-09-29:
Anyone? Please help