osqzss/gps-sdr-sim #267: Power level threeshold.

hosam37r, 2020-10-24:
Hello everyone. 
I was able to successfully spoof GPS on a USB GPS chip
But no luck on a galaxy s6 (even when no sim card and on airplane mode). One thing is suggested that android will throw away measurement if above certain threshold. 
So the question is What power level of GPS should be received on a phone.
Thanks,
HA

Mictronics, 2020-10-25:
Specified for GPS is a guaranteed reception power level of -136dBm on every place on earth.

Imtiaz08, 2022-01-03:
> Hello everyone. I was able to successfully spoof GPS on a USB GPS chip But no luck on a galaxy s6 (even when no sim card and on airplane mode). One thing is suggested that android will throw away measurement if above a certain threshold. So the question is What power level of GPS should be received on a phone. Thanks, HA

It is not about the power level of the GPS it is the receiver configuration of the phone. Since most phones use A-GNSS which include GLONASS, Galileo, BaiDou and GPS therefore, it will not be easy to spoof a multi constellation GNSS receiver with just GPS signal