osqzss/gps-sdr-sim #274: Why Hackrf not working on ios devices

Diagru25, 2020-12-28:
Hello everyone,
I have a problem about location simulation on iphone:
I use static mode and hackrf, it seems work very well on androi devices but ios devices ( iphone X, 11, ...) don't work
I have used lastest brdc file and tested -t/-T flag but it still not working. It just work on iphone < X (6,7,8)
What is my problem ?
How can I fix it ?
Thanks.

phields, 2021-01-29:
same issues

Diagru25, 2021-01-30:
I still haven't found that problem :(. If you know something, share it with me, pls.

Holmes-Shelly, 2021-02-07:
My hackrf works well on old devices like iphone se(2016), and T1(android phone in 2015). But, it seems that it can't cheat any new devices like iphone 11, android 10.

atchyuth-rao, 2021-05-07:
i also tested with mobile phones but i am unable to spoof on mobile phones,

i am using pluto sdr to generate GPS signal, any idea what sequence of operations should i follow to spoof basic phones

any idea please help me thanks in advance

vxling, 2022-01-24:
It works with: iphone <= 8/plus, and ios must <= 14.3

vxling, 2022-02-23:
works with iPhone 8 and iOS 14.4

IvanKor, 2022-02-23:
> i am unable to spoof on mobile phones

spoof functions not implemented in gps-sdr-sim 
if add spoof functions in gps-sdr-sim 
then 
radius spoof HackRF ~ 30 km 
radius spoof HackRF+PA_1W ~ 200 km

vxling, 2023-04-19:
Try multi-sdr-gps-sim,with a realtime mode

moheshmohan, 2023-05-15:
guys modern android and ios devices are really hard to spoof these days with this method so there is no point in trying. First these devices changed their behaviour with gps completly. i can confirm that i was able to successfully spoof both and android and ios in 2018 - 2019 but then when i try same steps in 2022 - 2023 they no longer work. I belive its because of below reasons

Phones no longer rely only on gps, they get location data from various sources like isp and wifi
Phones do not trust one source of gps they connect to multiple gps satellites and i am no expert but i think this method is only spoof on one type of source.

i was able to spoof simple usb receivers easily and none of the android or ios phones nearby were spoofed.