osqzss/gps-sdr-sim #278: $GPRMC "Speed over ground" inconsistent

Tom-sp765, 2021-01-11:
When I transmit a gpssim.bin file built with this software, the "speed over ground" in the $GPRMC messages output by the receiver I am using to record the transmission is always inconsistent (much slower than, and seemingly uncorrelated) with the simulated movement.

Everything else works beautifully. Is this something I can fix by changing a parameter?

osqzss, 2021-01-11:
The unit of the speed over ground in the RMC message is knot.

Tom-sp765, 2021-01-11:
Thanks for replying -  if I simulate a journey in which mean speed calculated from the location and time data is 50 knots, I'm getting a mean "speed over ground" of between 0 and 3 knots, and apparently no correlation between the speed over ground and the simulated speed. It's not just less - it doesn't appear to be related.

I've noticed a similar issue with "course over ground" - it doesn't seem to relate to the simulated direction inferred from the location coordinates, in the same way it does for "real" data.