osqzss/gps-sdr-sim #282: Please help me understanding with the gpssim.c code

komalkumar6655, 2021-02-16:
hello, i am a student. I am trying to understand the gpssim.c code . Please any body help  me.....!!!!!

Mictronics, 2021-02-16:
Should we guess your problems?
Why don't you ask specific question related to the code?

komalkumar6655, 2021-02-16:
Thank you.
I am having trouble in understanfing the generation of navigation message from ephemris data.
Pleae help me...

komalkumar6655, 2021-02-17:
[image]

komalkumar6655, 2021-02-17:
Can anybody expalin the subframe 1 in the above picture. How subframe 1 is generated.
thankyou.

osqzss, 2021-02-17:
You can find all the definitions of the message structure and contents in IS-GPS-200.
https://www.gps.gov/technical/icwg/

komalkumar6655, 2021-02-17:
Thank you for response. 
Can you expalin this 0X8B0000UL<<6 mean .
please.......

osqzss, 2021-02-17:
It's a preamble.
[image]

komalkumar6655, 2021-02-17:
Thanks
But why are you shifting it 6 times

komalkumar6655, 2021-02-17:
Please Explain this 
I will be Thankful to you.

ongbuntu, 2021-02-17:
> Please Explain this 
> I will be Thankful to you.
> 

Each word is 30bits long. 0x8B is the hex representation of the premble. By shifting 6bits to the left, you would put the premble in the right bits position withing the TLM word.

ongbuntu, 2021-02-17:
To learn the code, I strongly suggest that you first read the ICD and understand how each word is packed and what the individual parameters mean. Once you are able to understand one word, the rest of the code will start to flow.

komalkumar6655, 2021-02-17:
Thank you for your response..
But why to shift 6 times as the preamble is 0X8B == 10001011 .
Why to shift it 6 times . 
there is no information about this in the given document .
Please explain 
I will be very thankful.

komalkumar6655, 2021-02-17:
Please any one respond to the above comment.
Please......

Mictronics, 2021-02-17:
See the comment above from ongbuntu. In addition of aligning the preamble on MSB with <<6 you make up the space for the 6 bit parity on the LSB (left).

komalkumar6655, 2021-02-17:
Sorry I didnot understand .
can You please explain again..

k4cle, 2021-02-17:
Hello Tak,
I hope you and all yours are staying well during this Covid plague.  Long time no see!
All good here - just got my second vaccine shot so maybe I can get out and go to a restaurant or something before long!
Take care, hope to see you at one of the ION shows or some GPS event before long.
Doug BakerGPS Creations

https://www.gps.gov/technical/icwg/—

Mictronics, 2021-02-17:
@komalkumar6655 

> Can you expalin this 0X8B0000UL<<6 mean .

First look at the constant 0x8B0000UL. You will notice it's not a 32bit representation but 24bit (6 digits not 8), so in 32bit that's 0x008B0000UL.
```
0x008B0000UL
0000 0000 1000 1011 0000 0000 0000 0000
<< 6
xx10 0010 1100 0000 0000 0000 00pp pppp <  p = 6 parity bits

The entire TLM word is only 30 bits long (24 data + 6 parity).
So the upper two MSB (xx) are not relevant.

After <<6 the preamble 8B is MSB left aligned as required in spec.
```

komalkumar6655, 2021-02-17:
Thank you very much for your kind response.

Mictronics, 2021-02-17:
Did you ever read the IS-GPS-200 that was mentioned in a previous post?

(Ignore the 0 above word 1, edit mistake)
[image]
WN is in word 3 of subframe 1, bit position 71 in the **entire frame**.
The pink numbers in my paste are the word length and LSB in code. You see that LSB bit 1 of word 3 corresponds to bit 91 in subframe 1. Hence you need to shift <<20 (91-71=20) to place "wn" at the correct bit position in word 3 and therefore at the correct bit position in the entire frame.

PS: Maybe you should change your student class for gardening...

komalkumar6655, 2021-02-18:
since you have already studied the code and designed a new repository for multiple GPS SDR SIM.
I  am new to this GPS simulator and i didnot understand the logic behind it . Thats why I have asked many times about that.

closed 2021-02-18 by komalkumar6655