osqzss/gps-sdr-sim #283: Replay leap notification & Event

AndyCapp69, 2021-02-17:
I have seen some older posts on this, but what I would like to do is replay the leap notification from the day of 2016-12-31, then the leap event itself and then 15 minutes or so after the event.  I simply want to verify a GPS receiver handles leap seconds correctly.

I have downloaded some brdc files, but how do I combine 2016-12-31_23:00:00 with 2017-01-01_00:00:00 to form a continuous data stream for my HackRF?

Second, even generating just 15 minutes from brdc results in 4.7Gb - reducing MAC_CHAN to 3 reduced the output file, so did using "-b 1", but the file is still huge.   I can't see how I can generate a data file to cover 2 hours.  How have people managed to simulate a whole day ?

Mictronics, 2021-02-17:
Still under development, but simulates on hackrf without intermediate file generation:
https://github.com/Mictronics/multi-sdr-gps-sim

RINEX needs to be gz compessed though.

AndyCapp69, 2021-02-17:
Yes, a solution to the huge file.  That's nice.  But a quick look at the code and I could not see Leap Seconds mentioned anywhere. I need to test leap seconds on my device

Mictronics, 2021-02-17:
And how does gps-sdr-sim differs in that case?

AndyCapp69, 2021-02-17:
This is my original question - how can I replay Leap Second notifications & Events ?
This post hints it was done https://github.com/osqzss/gps-sdr-sim/issues/42  The gps-sdr-sim code is easy changed to allow a command-line param to change the default 18.

Mictronics, 2021-02-17:
Default parameters from last event are hardcoded in eph2sbf:
```
    // TO DO: Specify scheduled leap seconds in command options
    // 2016/12/31 (Sat) -> WNlsf = 1929, DN = 7 (http://navigationservices.agi.com/GNSSWeb/)
    // Days are counted from 1 to 7 (Sunday is 1).
    wnlsf = 1929 % 256;
    dn = 7;
    dtlsf = 18; 
```
Delta time due to leap seconds (dtlsf ).
The week number (wnlsf) and the day number (dn) at the end of which the leap second becomes effective.

I guess to replay you need a appropriate RINEX file (with iono data, and leap second set to 17?) then replay with parameters given in #42.
e.g. `gps-sdr-sim -e brdc2270.16n -l 35.658581,139.745433,100 -T 2016/12/31,23:57:00`

Imtiaz08, 2022-01-05:
@Mictronics is the multi-gnss sdr ready now and can we use it on windows OS too?