osqzss/gps-sdr-sim #292: Two hours shift from choosen date and time

Floh-dr, 2021-06-13:
Hi,

Using this project to spoof phones, I noticed that the time displayed by the GPS application I use to show constellation of satellites (GPS Test) is always 2 hours later than the one that I set with the parameter -t (or -T). I've search a lot about RINEX format and if it could be the reason of this shift but I cannot fnd the answer.
Could it be because of the choosen region when initializing the phone (which is actually a zone corresponding to UTC-2 in my case) ?

Thank in advance !

DL7AD, 2021-06-21:
Yes GPS uses GPS time not local time.

Important Note: GPS Time is not UTC Time. It's close to UTC but offset by 18 seconds at the moment.

Floh-dr, 2021-06-21:
There was a mistake, I'm at UTC+2, not UTC-2.
Therefore I cannot understand the difference in time between the hour displayed by GPS Test and the one I choose for the simulation. Because the time in GPS Test is actually at UTC+4.

DL7AD, 2022-05-05:
I've used this feature just recently and I ran into the same error. If I recall correctly, you use a timestamp around  2014-12-20 it should work. But in my case my receiver always showed always a date around year 2034. No idea why.

If you maybe get a new RINEX file, it will maybe work? This one in the repository is fairly old.

MrChernik, 2022-05-08:
well... is it possible to specify the current time in command ?? i tried
this with iphone, so i almost broke it, when it started showing 2014
year)))