osqzss/gps-sdr-sim #293: USRP N210, no GPS fix

FuhrerKing33, 2021-07-13:
Greetings, 
I am trying to spoof my phone using USRP N210, i have installed the UHD and GNU radio on ubuntu, and have generated the latest signal file as well, but i am unable to send a spoofing signal or maybe my phone is not getting a fix. 
Is there a way for me to check if my gps signal is being transmitted successfully? 
[image]
I am using the following settings, i have tried changing the gain but to no avail, if anyone has a solution, I'd be thankful!

Ang-1997, 2021-07-15:
I spoofed my phone using USRP N210.It's abnormal that your terminal print so many 'U',it means PC not providing data quickly.What's more,it seems that you didn't use the command which the author provided.The following is the command i used.
cza@ubuntu:~$ ./gps-sdr-sim -e brdc1950.21n -l 39.9418406828,116.3422268626,100 -s 2500000 -b 8
cza@ubuntu:~$ sudo python gps-sdr-sim-uhd.py -t gpssim.bin -s 2500000 -x 20 -b 8
By the way**,it's very important that you should use "-b 8"**,because USRP N210 only support 8 bit file(maybe,at least i failed with default 16).At the same time,i suggested that you should use the python file gps-sdr-sim-uhd.py to transmit the signal like what i did.
With the below method,it‘s normal that there are several or none 'U' when you excuting the transmiting signal command.And to check if your gps signal is being transmitted successfully,you can use the "GPS Test" application on Android phone.The following pictures are the results of my experiment.
[image]

[image]

I'm not good at English,and i hope my comment can help you.

FuhrerKing33, 2021-07-16:
> I spoofed my phone using USRP N210.It's abnormal that your terminal print so many 'U',it means PC not providing data quickly.What's more,it seems that you didn't use the command which the author [...]

Thanks a lot for your response, I shall try out the changes you have suggested and let you know if it works.

Mask6asok, 2021-07-21:
> I spoofed my phone using USRP N210.It's abnormal that your terminal print so many 'U',it means PC not providing data quickly.What's more,it seems that you didn't use the command which the author [...]

Hello, i am using USRP b210 to do this.
I can see some satellites in GPS Test App, but it's GNSS Status is still "No Fix".
Do i need a GPSDO to solve this?

Ang-1997, 2021-07-21:
> Hello, i am using USRP b210 to do this.
> I can see some satellites in GPS Test App, but it's GNSS Status is still "No Fix".
> Do i need a GPSDO to solve this?

Actually,i faced this problem in my experiment too.Then i tried to get a recent .21n file on https://cddis.nasa.gov/archive/gnss/data/daily/.Then i succeeded with the new file.By the way,my USRP N210 doesn't have GPSDO.I don't sure B210 can also work like this.But i suggest you can try several files if it doesn't work.

Mask6asok, 2021-07-21:
> Actually,i faced this problem in my experiment too.Then i tried to get a recent .21n file on https://cddis.nasa.gov/archive/gnss/data/daily/.Then i succeeded with the new file.By the way,my USRP [...]

Thanks.

Imtiaz08, 2022-06-12:
Hi @

> I spoofed my phone using USRP N210.It's abnormal that your terminal print so many 'U',it means PC not providing data quickly.What's more,it seems that you didn't use the command which the author [...]

Hi, I tried testing the scenario with my USRP N210 (No GPSDO installed at the moment) as soon as I generate the signal it behaves as a jamming signal. Since I am using it in my LAB, therefore, I tried using the RTL SDR to test the incoming signal. But I wasn't even able to see anything on the RTL SDR too. 

Can you help me please? Also when I try to read the gpssim.bin file in GNSS SDR it says invalid preamble.

Imtiaz08, 2022-06-14:
Hi what antenna are you using with your USRP please?

Ang-1997, 2022-06-14:
I'm sorry for not replying in time.
     Firstly,I'm not sure what you mean，it seems that your USRP N210 works but your devices can't be spoofed?Actually,i didn't use GPSDO(I don't have GPSDO),so it's hard to let my phone to be spoofed,which made me try many times(about 30) and succeed once.I can't even do it a second time.The main reason is  the clock is not accurate enough.so I recommend you use GPSDO if you have one, this can increase the success rate of spoofing.
    Secondly,you can't see anything on the RTL SDR.you should confirm that if your USRP working properly or not.Because when i failed to spoof my phone,i still could see some signal on the GPS Test App,it just could't lock the signal.I suggest that you can download a GPS Test app on your phone,and test the signal when your USRP works.If you also can't get any signal,maybe you should focus on how to make your USRP work properly.By the way,you should test USRP indoor,or you will get the true satellite signal.
    Thirdly,i don't know much about the gpssim.bin file,so i send you my file for your test.
    The gps-sim.grc file is my gnuradio flow graph.
    To be honest,I'm not specialized in SDR,and i hope this can help you.

 

  
 

 
  
 
 

从QQ邮箱发来的超大附件

gpssim.bin (1.86G, 2022年07月14日 13:15 到期)进入下载页面：http://mail.qq.com/cgi-bin/ftnExs_download?k=59646137163eb7e9241c7e231033574e05055806020356544b070352541e515157534c0f0550014c5451510205510359525c0354363965061617125e5b1d070808645c&t=exs_ftn_download&code=fda763ea

Ang-1997, 2022-06-14:
I don't sure because i have graduated from my university.Maybe vert900.I think the antenna which can send 1575.42MHz signal is  ok.

Imtiaz08, 2022-06-14:
![Uploading d3{970fc61} (78af33fc-8ed6-4c44-afcb-d276686edf4d id=0x7f0a03c3 tag=BaseCommentFragment) …]

I am using this antenna for transmitting the signal

Imtiaz08, 2022-06-14:
[image]

Imtiaz08, 2022-06-14:
> 

If you can share your files it would be very helpful for me.

Imtiaz08, 2022-06-14:
@Ang-1997 Before transmitting the signal I could see some satellites with very weak SNR values in the GPS test app. After when I transmitted the signal all of a sudden GPS Test app showed me a loss of lock and the satellites disappeared. I tried covering the antenna and noticed that the satellites were back again. 

This means the signal is jamming the 1575.42 MHz frequency band and maybe due to the high gain value of my N210

Maybe if @Mictronics or @osqzss sensei could gave us some valuable hints in this regard

Ang-1997, 2022-06-14:
I think the antenna is ok,you'd better avoid interference from real satellite signals when you transmit the signal.Lastly, the GPSDO maybe can solve the the loss of lock.Good luck!

osqzss, 2022-06-14:
@Imtiaz08 You cannot use an active antenna to transmit the signal.

Imtiaz08, 2022-06-17:
The transmitted signal power is too low so can't lock satellites on Zed F9p ublox and USRP N210

osqzss, 2022-06-18:
GNSS active antennas are designed for signal reception, not for transmission.

Imtiaz08, 2022-06-19:
I am using wire transmission now but still, the receiver is not locking any satellite and the signals appear to be very weak for the lock in the U-Center software. I have tried receiving the signal using RTL SDR but still no luck. 

While studying the signal using Soft GNSS v3 (MATLAB) it says invalid preamble.

Imtiaz08, 2022-06-25:
I am receiving the signal on my phone but can't get a lock on to satellites as it appears and disappears suddenly. Also the power of the acquired satellites is too low.

osqzss, 2022-06-26:
How much attenuation did you apply between TX and RX line? You generally need around -50dB to obtain a moderate signal level for typical GNSS receivers. If the signal level is too high, the receiver will lock to whatever they see.

Imtiaz08, 2022-06-26:
I have connected the USRP directly to my Ublox C94 M8P receiver via cable. 

Besides that I've applied -x 20, 0, 75, 85 but couldn't get a lock. Although I can see satellites in my u center screen.

osqzss, 2022-06-26:
You need attenuation, not gain.