osqzss/gps-sdr-sim #296: What does "f_carr" represents in the ComputeCodePhase() function?

AntI-HI, 2021-09-07:
// Pseudorange rate.
rhorate = (rho1.range - chan->rho0.range)/dt;

// Carrier and code frequency.
chan->f_carr = -rhorate/LAMBDA_L1;
printf("f_carr -> %f\n", chan->f_carr);
chan->f_code = CODE_FREQ + chan->f_carr*CARR_TO_CODE;
printf("f_code -> %f\n", chan->f_code);
printf("chan->f_carr*CARR_TO_CODE -> %f\n", chan->f_carr*CARR_TO_CODE);

In the above code, based on the rhorate and wavelength of the L1 band, we determine the carrier frequency. But frequencies we found diverges tremendously from the 1575.42 khz. I get some values like 2870.789292, -2981.129475 etc. What is that negative frequency doing here? Can You explain the idea for this and the whole concept for that function?

terrymarkovich, 2021-09-08:
I believe that is just the doppler shift in Hz

Imtiaz08, 2022-01-02:
> 2870.789292 this much doppler shift?

terrymarkovich, 2022-01-02:
Yes, GPS doppler shift can be up to 5,000 Hz

Imtiaz08, 2022-01-02:
Oh! 

Can you also explain how is the spoofing signal generated using thr ephemeris and desired coordinates? I mean what parameters are changing and how is the İQ file generated?