osqzss/gps-sdr-sim #299: Pseudorange can't be calculated

yangfan852219770, 2021-09-25:
Satellites' signal can be received, but it can't calculate the pseudorange. What could be the reason?

Imtiaz08, 2022-01-02:
there could be multiple issues 

- SV clock error
- user clock error
- ephemeris parameters error

are you feeding the latest ephemeris?

pbingxu, 2022-02-15:
@Imtiaz08 , Hi there, in my experiment, I conducted a conductive test with u-blox receiver. In u-centre, pos. is pretty good. Then, I tested the observation file from u-blox using RTKLIB. Unfortunately, the single point positioning result is bad. I looked into the pseudorange, which is weird. Perhaps, as you mentioned, the clock terms have some issues.

Imtiaz08, 2022-02-15:
@pbingxu Yes, that's because the satellite position is being calculated wrongly due to the incorrect time which is resulting in a very abnormal pseudorange. 

You can try using the overtime TOE and TOC to the scenario time and then check the rhos calculated from the signal you will see a difference.

pbingxu, 2022-02-16:
@Imtiaz08 , hi Imtiaz, thank you for your reply. (1) I'm not sure why u-center works fine, but the RINEX file is not working. Attached is the RTKPOST results using SPP mode.  [image]

(2) From a receiver's perspective, the time for calculating satellite pos. is relying on the signals that are being processed. We find it's time of transmission, which is used for calculating satellite pos. I don't understand your solution, "using the overtime TOE and TOC to the scenario time", please elaborate more. Thank you very much!

Imtiaz08, 2022-02-16:
> working. Attached is the RTKPOST results using SPP mode. [image: RTKPOST
> Results (SPP)]

(2) when a receiver receives a signal it uses its own clock and the
satellite clock to find the time of transmission just as you mentioned.
Let's say the brdc3540.14n file was fed to the GNSS SIM and an IF signal
was generated from that file. Technically the signal will contain Toe and
Toc values as is in the RINEX file right?

While calculating the time of transmission you will see a huge difference
which will obviously calculate the position wrong and as a result, the user
position will have a very large pseudorange since rho = c*t. I hope you are
getting my point.

Now for question 1: Can you start receiving the spoofing signal and check
the pseudo-range values in the U-center, please?

Imtiaz08, 2022-02-16:
U-center takes current time for the calculation of signal transmission
time. While the signal contains toe and toc values depending on what the
RINEX file has

Imtiaz08, 2022-02-16:
Can you provide me with your coordinates? or at least tell me how much
difference are you getting?

pbingxu, 2022-02-16:
> 

Ground truth location: 22.3221720833333 114.141187963889 20.2386

pbingxu, 2022-02-16:
@Imtiaz08 , I will follow your suggestions. Will update you later. Thank you so much!