osqzss/gps-sdr-sim #305: Troubleshooting / Tutorial for gps-sdr-sim with HackRF and Android devices

JsBergbau, 2021-11-28:
**TLDR
TCXO recommended, then it works
Some (newer?) Android devices don’t get a fix, even this project works as expected**

Here are my experiences with gps-sdr-sim and HackRF. Cost me some time to get it work, so I want to share the experience here, so other users save a lot of time.

For HackRF simulation file created via `./gps-sdr-sim -b 8 -e brdc0170.21n -u circle.csv` worked. Replace the brdc with that one from the current day.

Transmitting the GPS signal was done via ` hackrf_transfer -t gpssim.bin -f 1575420000 -s 2600000 -a 1 -x 20` and a short dipole antenna of length 15 cm length in total. 
Transmitting worked, but the GPS-Receivers lost signal without showing a faked position. The frequency of this HackRF was obviously a bit mistuned. I’ve tried over one hour to get the right frequency, however only for a few seconds GPS receiver got a fix with the faked location. So I bought an TCXO, in particular it is the " NooElec Tiny TCXO: 0.5PPM TCXO Modul für HackRF" https://www.amazon.de/gp/product/B077H7Q7GL

For how to open the HackRF case, watch this video https://www.youtube.com/watch?v=zuXJtpTSEJM It is sufficient to loosen the screws on the side a bit. 

Take a look at this photo https://github.com/osqzss/gps-sdr-sim/blob/master/extclk/hackrf_tcxo.jpg how to place the TCXO.

After installing the TCXO use ` hackrf_debug -–si5351c -n 0 -r` to check if TCXO is recognized and used. Output ` [0] -> 0x01` means TCXO is used, whereas ` [0] -> 0x51` means no TCXO is used.

After transmitting via TCXO fake position worked on an older Android device and on a cheap U-Blox7 GPS-Dongle like https://www.amazon.de/DollaTek-Antenne-VK-172-Empfänger-Windows/dp/B07F6Y6DPT/ 
However on Android 11 on Galaxy S10 Lite there is always "No Fix", even though phone was put into airplane mode and also A-GPS Data was deleted. For all tests App "GPS-Test" https://play.google.com/store/apps/details?id=com.chartcross.gpstest&hl=de&gl=US was used.

Since this device uses also GlONASS, Beidou and Galileo first thought was that the reception of additional systems prevent the fix. So the phone was also brought to the basement where no GPS reception is possible. Even after 30 minutes of transmitting a fixed position in a loop Galaxy S10 Lite did not fix GPS position, whereas the other Android phone with Android 6.0 got a fix the whole time. I’m willing to debug to improve the code to fake also GPS signal on Galaxy S10 Lite / Android 11.

Imtiaz08, 2022-01-01:
Let me know if you need a hand with this project. I am also working on GNSS Spoofing as my final semester thesis. 

I am looking for someone to collaborate with

JsBergbau, 2022-01-01:
@Imtiaz08 
This isn't my project. Sadly I don't have enough technical and mathematical skills to help in developing this project. I'd be very happy if you can improve this project and I can help you with some testing stuff.
@osqzss I mention you so you can get in contact with @Imtiaz08

dalcala68, 2022-06-09:
Best regards,

Good evening my name is David Forero I am a PhD student at the university of Alcala. I am trying to use your software package to create orbital paths in low orbit. I create a csv file of ECEF positions at 10Hz and the following happens to me: with a GPS ublox Neo 6 it connects without problems and makes me the orbital path roughly well. But if I use a GPS NEO M8T the gps does not connect. For the project I am doing I have to use the GPS neo M8T and read its log which delivers the data in ECEF directly.

Please can you help me by giving me an indication of how to create this file as a multi-sdr-gps-sim collaborator to work with the GPS neo M8T.

Regards

thank you very much for your help 

David Eugenio Forero
dforero@radiogis.uis.edu.co

Imtiaz08, 2022-06-09:
Hi are you transmitting the file with HackRF and not able to receive the signal using Neo M8T?

dalcala68, 2022-06-09:
Hello

Yes. I can't receive it with the NEO M8T the gps won't connect. but my file
does work with an ublox Neo 7 GPS.

Using the software u centre from ublox all the bars are blue with the M8T.
Tomorrow I can send you a summary with pictures of what I am doing to see
if it is possible to find a solution.

This is a proof of concept for my doctoral thesis.

Greetings
Thank you very much for replying

David Forero

osqzss, 2022-06-09:
Neither NEO M8T nor NEO 7 is capable of providing navigation solutions of an LEO satellite.

dalcala68, 2022-06-10:
Good afternoon,

In reference to my question of yesterday in the attached files you will
find the Test 1 folder in it is the csv file and its equivalent binary,
generated by gpssim to transmit with the radio sdr, which builds my
application developed in Python. that as I mentioned yesterday with a GPS
ublox NEO 7 makes me a path of orbit well, but with negative height but
equivalent.
That same binary file does not work with the GPS ublox Neo M8T and presents
a performance as shown in the images and the attached recording or center.
It is worth mentioning that these files are generated with an application
that calculates the orbital path solving the two-body problem and
calculates a position in the ECI reference system and then it is
transformed to ECEF taking into account the effects of nutation and
precession of the earth and the epoch referred to the Julian day calculated
at all times.

In folder 2 my Python function does not take nutation and precession
effects to generate the files. First it generates an ECI file and
transforms them WGS 84 and then transforms it to ECEF the behavior is the
same: it connects to the Neo 7 but with the Neo M8T it does not work, the
bars are dark blue and light blue at other times and it does not process
the information.
I attach my Python codes to see if you can do me the favor of giving me
help.

thank you very much
Regards

GPS are not spatial level and have speed and height restrictions the idea
is just to make a proof of concept.

David Forero

https://drive.google.com/file/d/1CxS-a34s8wRgNLKoVspjC3DdHROcy0B5/view?usp=sharing

osqzss, 2022-06-10:
Under the Wassenaar Arrangement, GNSS receivers are supposed to disable themselves when they detect that they are above 18,000 m **and** moving faster than 1,900 km/h.

dooonabe, 2022-07-08:
@JsBergbau i get three android phones: redmi4(android6.0), redmi6(android8.0),samsung s10(android12),only android6.0 can fix the gps, others can view the satellites but can't fix them.

dooonabe, 2022-08-29:
> @JsBergbau i get three android phones: redmi4(android6.0), redmi6(android8.0),samsung s10(android12),only android6.0 can fix the gps, others can view the satellites but can't fix them.

Wow! Recently ,I test on redmi6(android8.0) and vivo y67(android8.0), they can fix the satellites, i just updated the [hackrf-2021.03.1.zip](https://github.com/greatscottgadgets/hackrf/releases/download/v2021.03.1/hackrf-2021.03.1.zip) without maked any other changes

Commander-Crash, 2023-03-29:
My 1st Galaxy Note10 will not lock. running android 12. My 2nd galaxy note 8 will lock all day long. I know this is not helping but i thought i would confirm this myself.