osqzss/gps-sdr-sim #307: Calculated satellite position does not match with real position

knrl, 2021-12-27:
Hi,

I want to check lat, long, alt information of satellites. When I converted the pos[0], pos[1], pos[2] values in the computeRange() function to lla and compared them with the values from in-the-sky.org/satmap_worldmap.php, I noticed that the lat, long, alt did not match. I checked this for the same prn, toe and date values. Is there something I've missed? I would be glad if you inform me about this.

Imtiaz08, 2022-01-01:
Hi there, can you provide some details on what you are actually doing? I am a GNSS student and I think I can help you further

knrl, 2022-01-02:
I want to check the correctness of the satellite position calculated by gps-sdr-sim. Since gps-sdr-sim calculates the position from the ephemeris parameters in the rinex file, these positions must match the satellite's position information for the date the rinex file has. The position of the satellite in a certain time interval can be seen on In-the-sky.org.

Imtiaz08, 2022-01-02:
Actually, I am coding in a Matlab environment so I won't be able to help much in other languages but I think you can get a rough idea of calculating the satpos from the given ephemeris file, 

` GM = 3.986005e14;
    %Rotation of earth (omega_dot)
    Omegae_dot = 7.2921151467e-5;
    for lo = 1:32
    % Assigning local variables to the ephemeris
    svprn   =   ephemeris(1,lo);
    af2     =   ephemeris(2,lo);
    M0      =   ephemeris(3,lo);
    roota   =   ephemeris(4,lo);
    deltan  =   ephemeris(5,lo);
    ecc     =   ephemeris(6,lo);
    omega   =   ephemeris(7,lo);
    cuc     =   ephemeris(8,lo);
    cus     =   ephemeris(9,lo);
    crc     =  ephemeris(10,lo);
    crs     =  ephemeris(11,lo);
    i0      =  ephemeris(12,lo);
    idot    =  ephemeris(13,lo);
    cic     =  ephemeris(14,lo);
    cis     =  ephemeris(15,lo);
    Omega0  =  ephemeris(16,lo);
    Omegadot=  ephemeris(17,lo);
    toe     =  ephemeris(18,lo);
    af0     =  ephemeris(19,lo);
    af1     =  ephemeris(20,lo);
    toc     =  ephemeris(21,lo);

    %Calculate SV Position at Every Epoch 
        A = roota*roota;
        tk = check_t(t-toe);
        n0 = sqrt(GM/A^3);
        n = n0+deltan;
        M = M0+n*tk;
        M = rem(M+2*pi,2*pi);
        E = M;
        for i = 1:10
            E_old = E;
            E = M+ecc*sin(E);
            dE = rem(E-E_old,2*pi);
            if abs(dE) < 1.e-12
            break;
            end
        end
        E = rem(E+2*pi,2*pi);
        v = atan2(sqrt(1-ecc^2)*sin(E), cos(E)-ecc);
        phi = v+omega;
        phi = rem(phi,2*pi);
        u = phi              + cuc*cos(2*phi)+cus*sin(2*phi);
        r = A*(1-ecc*cos(E)) + crc*cos(2*phi)+crs*sin(2*phi);
        i = i0+idot*tk       + cic*cos(2*phi)+cis*sin(2*phi);
        Omega = Omega0+(Omegadot-Omegae_dot)*tk-Omegae_dot*toe;
        Omega = rem(Omega+2*pi,2*pi);
        x1 = cos(u)*r;
        y1 = sin(u)*r;
        satp(1,ep) = x1*cos(Omega)-y1*cos(i)*sin(Omega);
        satp(2,ep) = x1*sin(Omega)+y1*cos(i)*cos(Omega);
        satp(3,ep) = y1*sin(i);
    end  `

Imtiaz08, 2022-01-02:
However, I am interested to learn how is the simulator generating an IF file using the ephemeris and desired coordinates. Since I don't have a C++ background, therefore, it seems so confusing... I mean how can it embed the coordinates in the given rinex parameters what is actually happening in the background. 

can someone please explain the process with a flow chart or something?

knrl, 2022-01-02:
Thanks for your answer but I'm not asking how to calculate satellite position from ephemeris parameters. I'm asking why the calculated satellite positions in gps-sdr-sim do not match the in-the-sky.org positions. If the gps-sdr-sim location calculation is correct, there should be match with in-the-sky.org positions. Because both of them use the same ephemeris parameters.

IvanKor, 2022-01-03:
> I'm asking why the calculated satellite positions in gps-sdr-sim do not match the in-the-sky.org positions.

**all GPS functions (secret antispoofing) not implemented in gps-sdr-sim** 
this is especially noticeable in the u-center 
if observed **pseudorange residual**

Imtiaz08, 2022-05-20:
@knrl did you find an answer to your question?

jnunezortuno, 2022-06-02:
> @knrl did you find an answer to your question?

Any solution? I get an erroneous simulated constellations compared with the ones I get from http://www.taroz.net/GNSS-Radar.html

Imtiaz08, 2022-06-02:
@jnunezortuno how are you reading the signal? 

Are you transmitting it or reading it directly via some software?

jnunezortuno, 2022-06-09:
I am testing which satellites are visible to gps-sdr-sim at a specific time and comparing them with the real ones.
The results are as follows. For a simulation at position LLH [0,0,0] and date/time 2022/06/09, 10:00:00.

**GPS-SDR-SIM**
PRN AZ EL    
02 225.4 37.4    
05 334.6 44.2    
06 171.7 23.9   
11 216.1 43.5   
12 208.9 13.8   
13 344.0 18.5 
14 48.6 37.9   
15 316.3 0.4    
19 144.0 22.0   
20 334.6 78.8    
24 262.5 18.2   
28 68.4 59.2    
30 29.3 9.3    

In the image you can see the constellation provided by GNSS-Radar (http://taroz.net/GNSS-Radar.html) or Trimble GNN planning online (https://www.gnssplanning.com/) with the same conditions.

[image]

As you can see,  there is no agreement in the azimuth and elevation values, and even in the PRN of the visible satellites.

Please, can someone tell me what's going on?