osqzss/gps-sdr-sim #309: No PRN acquired after feeding the bin file to GNSS sdr

Imtiaz08, 2022-01-04:
Hi I tried generating the spoofing signal for study purposes and when I try to read it through GNSS SDR in Matlab it acquires no satellites. 

The following are my commands please let me know if I am doing something wrong... I also tried feeding the latest ephemeris file from brdcn but technically if it is generating a spoofing signal then it should generate a signal irrespective of the ephemeris date... 

[image]

[image]

komalkumar6655, 2022-01-05:
Hello I am also intrested in GNSS spoofing. I think the GNSS SDR matlab GUI settings should be changed. In the GUI settings replace the acquisition metric from 2 to 1.5

komalkumar6655, 2022-01-05:
And also try by the given example command in the original repository

Imtiaz08, 2022-01-06:
it is good to know that we both share common interests... Actually, I am working on a journal paper and doing this research in a controlled environment with real-time transmission. 

Have you got any luck with learning the c code? I don't understand how is the software embedding the given lat and long in the signal...

komalkumar6655, 2022-01-06:
Yeah I understood the overview of the Simulator just a bit. Do You want to share?

Imtiaz08, 2022-01-06:
thank you it worked

*From: *komalkumar6655 ***@***.***>
*Sent: *Wednesday, January 5, 2022 9:18 PM
*To: *osqzss/gps-sdr-sim ***@***.***>
*Cc: *Imtiaz Nabi ***@***.***>; Author
***@***.***>
*Subject: *Re: [osqzss/gps-sdr-sim] No PRN acquired after feeding the bin
file to GNSS sdr (Issue #309)

And also try by the given example command in the original repository

—
or Android

Imtiaz08, 2022-01-06:
> Yeah I understood the overview of the Simulator just a bit. Do You want to share?

Yes, It would be helpful for both of us... do you know how does it generates the signal based on our desired coordinates and ephemeris file?

komalkumar6655, 2022-01-06:
yes Let us understand the topic called pseudorange. From The ephemris file we calculates the satellite positions. THe desired location is also converted to xyz coordinates. satllite position (xyz)- desired position xyz gives the pseudorange. THen the pseudorange is converted to codephase which is used on generating the C/A codes. this is the overview. Now Coming to the receiving part the code phase is converted to pseudorange and then the pseudorange and the satellite psoitions are used to calculate the Receiver position by Least square algorithm which is the Desired location.

komalkumar6655, 2022-01-06:
I hope you understood about the overview . If not ping me.

Imtiaz08, 2022-01-06:
So you mean whichever position I want to achieve from the signal I need to generate its code phase and pack it in the signal?

komalkumar6655, 2022-01-06:
yeah absolutely correct. you got it!!!!!!!!!!!!!

Imtiaz08, 2022-01-07:
Amazing... do you have any paper related to this simulator that explains the steps etc in details? Actually I am learning how to convert rho to code phase and do we need to generate the whole signal i.e. subframes and nav messages?

komalkumar6655, 2022-01-07:
Yes You can check my paper regarding the GPS spoofing from the below link
https://ieeexplore.ieee.org/document/9397980

Imtiaz08, 2022-01-07:
Mr. @komalkumar6655 I can not thank you enough for sending me your research paper. I have just printed it and will read it thoroughly. I am sure it will be helpful in my research. 

One more thing... I was wondering if we can implement such a spoofer in a Matlab/Simulink environment... is it possible? do you think Matlab is capable enough to generate a GNSS signal? basically, I am trying to extend this approach to GPS+Galileo signals... 

can I contact you by mail or something?

komalkumar6655, 2022-01-07:
Are you expert in matlab because i need a hand for implementing the Spoofer in Matlab. Yeah Matlab is capable for generating  the GPS spoofed Signals. You can contact by mail: **skkumar96@outlook.com**

Imtiaz08, 2022-01-07:
I wouldn't call myself an expert but I can do pretty much things in Matlab and can read a written code in Matlab

komalkumar6655, 2022-01-07:
Okay Dou you help me in handlong the project of the IRNSS Simulator.

komalkumar6655, 2022-01-07:
U can contact me through the mail id i have given before

Imtiaz08, 2022-01-07:
Please check your mail/spam folder whenever you have time I have already mailed you

danziggit, 2022-08-22:
Hi, I am very interested in using GPS-SDR-SIM and GNSS SDR (Matlab scripts) to assist in eventually develop robust GPS C/A spoof detection. Please let me know if are two are still available to collaborate. Thanks.

Imtiaz08, 2022-08-22:
@danziggit I have recently completed my Thesis on GNSS Spoofing and now I am about to start working on anti spoofing. I am open for collaborations though

danziggit, 2022-08-24:
Hi @Imtiaz08 , congratulations on completing your thesis on GNSS Spoofing! I am starting my thesis on the same topic by applying deep learning for predicting if a receiver is being spoofed. For the past few days/weeks, I have been trying to use SoftGNSS to determine a Nav solution. Everything appears to work just fine (data looks good and +4 satellites are tracked) until I run "postNavigation.m". At that point I get an error stating the preambles cannot be found for any of the tracked satellites. The data I am using various ~60 second recording from an RTL-SDR (RTL2832 R860 with TCXO and Bias-T), and generated binary files from using gps-sdr-sim and Pothos SDR. Please let me know if you know of a solution or if you have ran into a similar problem. Thank you very much :)

Imtiaz08, 2022-08-24:
During my research when I was trying to run the SoftGNSS for the verification of the baseband signal I was getting the same error. 

There seems to be 2 major problems 
- The user is allowing too much noise in the loops
- The signal may not be sampled correctly.

However due to shortage of time, I choose to avoid inspecting the issue for the time being and use an alternative software called "GNSS SDR" which works in Ubuntu and it is pretty straightforward to use. 

I have now started to investigate the softGNSS and will share the solution as I find one

danziggit, 2022-08-24:
Here is the error I am receiving: 
   **Tracking started at 23-Aug-2022 22:29:31
   Tracking is over (elapsed time 00:00:36)
   Saving Acq & Tracking results to file "trackingResults.mat"
   Calculating navigation solutions...
Could not find valid preambles in channel 1!
Could not find valid preambles in channel 2!
Could not find valid preambles in channel 3!
Could not find valid preambles in channel 4!
Could not find valid preambles in channel 6!
Could not find valid preambles in channel 7!
Unrecognized field name "IODC".
Error in postNavigation (line 97)
    if (isempty(eph(trackResults(channelNr).PRN).IODC) || ...
Error in postProcessing (line 146)
    navSolutions = postNavigation(trackResults, settings);
Error in init (line 75)
    postProcessing**

Imtiaz08, 2022-08-24:
Please increase the data logging duration and make it at least 2 minutes.

danziggit, 2022-08-24:
Thank you very much for the suggestion. I generated more data longer than 200 seconds and ran into additional issues. I am going to take your advise and use GNSS SDR on Ubuntu.