osqzss/gps-sdr-sim #314: Can any body explain the following code section please?

Imtiaz08, 2022-02-02:
`tatic void computeCodePhase(channel_t *chan, range_t rho1, double dt) {
    double ms;
    int ims;
    double rhorate;

    // Pseudorange rate.
    rhorate = (rho1.range - chan->rho0.range) / dt;

    // Carrier and code frequency.
    chan->f_carr = -rhorate / LAMBDA_L1;
    chan->f_code = CODE_FREQ + chan->f_carr*CARR_TO_CODE;

    // Initial code phase and data bit counters.
    ms = ((subGpsTime(chan->rho0.g, chan->g0) + 6.0) - chan->rho0.range / SPEED_OF_LIGHT)*1000.0;

    ims = (int) ms;
    chan->code_phase = (ms - (double) ims) * CA_SEQ_LEN; // in chip

    chan->iword = ims / 600; // 1 word = 30 bits = 600 ms
    ims -= chan->iword * 600;

    chan->ibit = ims / 20; // 1 bit = 20 code = 20 ms
    ims -= chan->ibit * 20;

    chan->icode = ims; // 1 code = 1 ms

    chan->codeCA = chan->ca[(int) chan->code_phase]*2 - 1;
    chan->dataBit = (int) ((chan->dwrd[chan->iword]>>(29 - chan->ibit)) & 0x1UL)*2 - 1;

    // Save current pseudorange
    chan->rho0 = rho1;
}`
in the upper code the rho1 is given as input but this variable isn't declared anywhere else.

IvanKor, 2022-02-02:
> }` in the upper code the **rho1 is given as input but this variable isn't declared anywhere else**.

look 
```
				// Update code phase and data bit counters
				computeCodePhase(&chan[i], rho, 0.1);

```

Imtiaz08, 2022-02-02:
Couldn't find this line in the C Code but I will have another look and see if I can find it. 

Also 

I am confused about the time in the following code there is g0 which seems to be GPS time from the rinex file right? what is g1? is the current time? I mean the time at which we want to create the signal?

````
double subGpsTime(gpstime_t g1, gpstime_t g0)
{
	double dt;
	dt = g1.sec - g0.sec;
	dt += (double)(g1.week - g0.week) * SECONDS_IN_WEEK;

	return (dt);
}
gpstime_t incGpsTime(gpstime_t g0, double dt)
{
	gpstime_t g1;
	g1.week = g0.week;
	g1.sec = g0.sec + dt;

	g1.sec = round(g1.sec * 1000.0) / 1000.0; // Avoid rounding error
	while (g1.sec >= SECONDS_IN_WEEK)
	{
		g1.sec -= SECONDS_IN_WEEK;
		g1.week++;
	}
	while (g1.sec < 0.0)
	{
		g1.sec += SECONDS_IN_WEEK;
		g1.week--;
	}
	return (g1);
}`
```

Imtiaz08, 2022-02-02:
> > }` in the upper code the **rho1 is given as input but this variable isn't declared anywhere else**.
> 
> look
> 
> ```
> 				// Update code phase and data bit counters
> 				computeCodePhase(&chan[i], rho, 0.1);
> ```

found these lines in another code multi sdr gps sim but couldn't find it in the gps sdr sim...

IvanKor, 2022-02-02:
> Couldn't find this line in the C Code

https://github.com/osqzss/gps-sdr-sim/blob/master/gpssim.c

line  2174

Imtiaz08, 2022-02-02:
Can't understand what exactly is happening here... What is the time here? User time at which the signal Will be generated? Or the year month day time which was converted to gpst from the rinex file?

Imtiaz08, 2022-02-02:
İf possible can you please explain the computecodephase?