osqzss/gps-sdr-sim #318: Where is the dwrd being calculated and what does it represents?

Imtiaz08, 2022-02-27:
Hi there's a variable called Dwrd which stores an array of 60 elements... Can anybody tell me what does it represents please? And where is it being calculated? 

It is so confusing please anybody???

shuwang1, 2022-05-11:
I guess you know the answer already.

Imtiaz08, 2022-05-12:
> I guess you know the answer already.

Yes, I've learned but I still don't understand when we have subframes already then what is the purpose of DWORD generation? 

I have generated a signal in Matlab but when I try to see it in the GNSS SDR it says couldn't find a valid preamble... I even tried generating the signal using GPS-SDR-SIM and the result was the same...