osqzss/gps-sdr-sim #320: The code generates only 1000 PRN code chips however, the actual code is 1023 chips long why is that so ?

Imtiaz08, 2022-04-04:


Imtiaz08, 2022-04-05:
@scateu @edy555 @sgk @laf0rge @Mictronics @osqzss

osqzss, 2022-04-05:
The C/A code length is defined  as CA_SEQ_LEN in gpssim.h:

/*! \brief C/A code sequence length */
#define CA_SEQ_LEN (1023)

Imtiaz08, 2022-04-05:
Yes, but when I am debugging the code there are only 1000 elements only

Also can you please tell me why are you taking only subframe 5 and subframe 1 to generate the Dwrd in the GenerateNavMessage function?

Imtiaz08, 2022-04-05:
[image]

osqzss, 2022-04-05:
https://stackoverflow.com/questions/57921875/debugging-in-vscode-watching-vectors-with-length-1000

Imtiaz08, 2022-04-05:
Thank you so much sensi can you please answer the second question? i.e. why are you taking only subframe 5 and subframe 1 whole generating the navigation message? Also, are you calculating all the values in decimal? because again in the debugger we can see the decimal values only

osqzss, 2022-04-05:
Please ask Google first.
https://stackoverflow.com/questions/39973214/how-to-display-hex-value-in-watch-in-vscode

osqzss, 2022-04-05:
All the subframes are updated when generateNavMsg() is called:

int generateNavMsg(gpstime_t g, channel_t *chan, int init)
{
  ...
  for (isbf=0; isbf<N_SBF; isbf++)
  {
    // update subframes here.
    ....
  }
}

Imtiaz08, 2022-04-05:
so basically the eph2sub function converts ephemeris into the subframe and the generatenavMsg function updates the message right?

And I am sorry for not making myself clear enough... Actually, I am implementing the code in Matlab for my MS thesis so I would like to know whether the  navigation message is being generated in binary format or it is converted into decimal before writing it to bin file

Imtiaz08, 2022-04-08:
@osqzss sensi, I am using Matlab to write the generated binary data into a file. Can you please provide me any insights on this?

osqzss, 2022-04-08:
https://www.mathworks.com/help/satcom/ug/gps-waveform-generation.html

shuwang1, 2022-05-11:
> so basically the eph2sub function converts ephemeris into the subframe and the generatenavMsg function updates the message right?
>
> And I am sorry for not making myself clear enough... Actually, I am [...]

I think they are always in binary, not in decimal, because binary operations, e.g, checksum, are involved.

Imtiaz08, 2022-05-12:
@shuwang1 But the values are always shown in decimal when using VS code