osqzss/gps-sdr-sim #321: amplitude -inf dBfs

AFLAH-7jri, 2022-04-09:
iam using hackrf one and i used static mode. 
when i started transmitting the TX led light on hackrf is red and it shows me in cmd:
 5.0 MiB / 1.001 sec =  5.0 MiB/second, amplitude -inf dBfs
 5.2 MiB / 1.005 sec =  5.2 MiB/second, amplitude -inf dBfs
 5.5 MiB / 1.018 sec =  5.4 MiB/second, amplitude -inf dBfs
 5.2 MiB / 1.010 sec =  5.2 MiB/second, amplitude -inf dBfs
 5.0 MiB / 1.003 sec =  5.0 MiB/second, amplitude -inf dBfs

what does  "amplitude -inf dBfs" means ?

AFLAH-7jri, 2022-04-11:
[image]

saffetblt, 2023-01-04:
is there any progress?

AFLAH-7jri, 2023-01-04:
Not yet

CongQing-0719, 2024-03-31:
is there any progress?

osqzss, 2024-03-31:
https://github.com/greatscottgadgets/hackrf/issues/1121

ardyitjen, 2024-04-19:
is there any update? i got sampe problem with portapack h2