osqzss/gps-sdr-sim #323: I want to generate QZSS(prn193~199).

kurokkasu, 2022-05-09:
QZSS is compatible with GPS.
In creating it, the current changes are the generation of C / A code and the reading of the RINEX Navigation file(2.12).
Now the system is working, but not receiving.
What else do I need to change?
Please give me advice.

shuwang1, 2022-05-11:
"not receiving?" what does this mean?

kurokkasu, 2022-05-12:
The signal is being sent but the data cannot be obtained.
[image]

[image]

Imtiaz08, 2022-05-20:
There could be multiple reasons 

1. Check whether the preamble is valid in the DWRD 
2. Make sure you are correctly simulating the doppler frequency shift 
3. Make sure you are transmitting enough subframes
4. Make sure that the receiving module is not aided by anything such as A-GPS, SBAS etc and if it is mobile phone then you will need to turn off the Wifi/mobile data too

Imtiaz08, 2022-05-20:
Also the SVs in the sky plot and the SVs in the signal section seems to be different it means you are mixing stuff as far as I suspect

kurokkasu, 2022-06-07:
I recently re-received.
[image]
Subframe and DWRD debug results
[image]
[image]
[image]
How can I check the correct subframe?

kurokkasu, 2022-06-07:
I recently re-received.
[image]
Subframe and DWRD debug results.
![qzssdwrd](https://user-images.githubusercontent.com/82863897/172289115-135486[image]
76-a31c-4310-8086-88ab36627c61.PNG)[image]
How can I see the correct subframe?

Imtiaz08, 2022-06-07:
Subframe data and the dwrd data are different

Dwrd is the message generated from the subframes

osqzss, 2022-06-07:
Use a u-blox receiver. UBX-RXM-SFRBX message contains subframe data bits.

Imtiaz08, 2022-06-07:
@osqzss when I transmit the data I can see satellites visible but I can't get any fix. I've tried reading the file using GNSS SDR but it says invalud preamble.

kurokkasu, 2022-06-07:
@Imtiaz08 Is it the codegen that needs to be changed?

Imtiaz08, 2022-06-07:
No it is absolutely correct. You need to understand what's happening. Debug the code line by line and you will be able to see

kurokkasu, 2022-06-07:
@osqzss Is any kind okay?

kurokkasu, 2022-06-07:
@Imtiaz08 I understand, I'll do my best

Imtiaz08, 2022-06-07:
> @Imtiaz08 I understand, I'll do my best

imtiaznabi8@gmail.com this is my email. You may reach out to me if you want to work this project together