osqzss/gps-sdr-sim #326: Invalid preamble

Imtiaz08, 2022-06-07:
Hi, I have generated the signal file but when I try to read it through GNSS SDR (MATLAB) it says invalid preamble and there are only 4 satellites in the acquisition. 

Can anyone guide me if I am doing something wrong, please?

[image]

here's the error that I am currently getting 

[image]

figures 

[image]

jmsigler, 2022-07-20:
Hi Imtiaz, if you didn't already realize this, you're not actually locked to the satellites. While acquisition produced results, it looks like the signal strength was too low to actually lock onto the satellites (thus no data and no preample). You can tell this just by looking at the I/Q plot which should have two circles not one or the "bits of the navigation message" which, as you can see, contains no bits. I would try decreasing your attenuation and regenerating the file. Otherwise it seems like the file may not have been generated correctly.

osqzss, 2022-08-28:
This should be fixed in the latest code.

Imtiaz08, 2022-08-28:
We can work on this together if you can supervise me

osqzss, 2023-06-11:
Please use "-b 8" option to generate int8 I/Q samples.