osqzss/gps-sdr-sim #327: difference between Nav messages in UBX-SFRBX and generated ones

marcospanghero, 2022-06-10:
Hi,
I am looking at the generated NAV messages from the simulator and the received ones at the receiver. While frame 4 and 5 are correct and consistent between simulator and receiver, subframes 1 and 2 are different between the received ones and the simulated ones.

Am I missing something?

marcospanghero, 2022-06-10:
Also, simulated nav messages are very different than real ones for the same period of time (so I assume using the same eph data). i.e. word 1 for the same satellites in L1C is very different

Imtiaz08, 2022-06-14:
@marcospanghero did you transmitted the signal over the air?

I am using USRP N210 and my signal turns into jamming. I am not getting any fix

marcospanghero, 2022-09-29:
Over cable. OTA you need to be really close. Most likely the clock in the radio is not good enough. Get a good GPSDO or OCXO and use the external clock of the USRP