osqzss/gps-sdr-sim #328: required input data

abbasi518, 2022-06-19:
I want to know whether this code requires the actual coordinates and position of the object/receiver that is going to be spoofed as input or not. In other way, does this code uses the signals of satellites above this object/receiver for spoofing or the ones that are positioned on the spoof-trajectory?

Imtiaz08, 2022-06-19:
Hi @abbasi518 

The code is actually not built for spoofing and it is more of a simulator to test GPS receiver. However, the baseband signals generated by this code can be transmitted through an appropriate SDR setup to fool GPS receivers out there. 

Coming back to your question, it actually takes a RINEX file that contains the ephemeris data for satellites (you can either get it from the NASA's website or record your own using GPS receivers)

It doesn't require actual coordinates but it needs a target location to simulate the visible satellites for that location, their doppler frequency shift, and their code phases. 

No, it doesn't use the signals of satellites above your victim device (that you want to spoof). It is possible that the satellites produced in each channel created by this software are different from the ones that appear directly on the target location. However, it still does its job perfectly. 

This software just guesses the available satellite on the target position from the elevation angles calculated for each satellite in the whole constellation. 

Please feel free if you want to ask anything else. 

Thank you

Gerald0217, 2023-11-15:
"Hey there expert, I wanted to ask, does this simulated GPS signal depend on the ephemeris file? The ephemeris file gets updated every day, so every time I simulate, I have to go to the website to download the current ephemeris file. Is there a way to generate today's ephemeris file through a program or some other method? Another thing, you mentioned that the simulated satellites for the current location may not actually be the real satellites above the target location, so the simulated GPS signal has a loophole. Can this issue be resolved? Are the simulated GPS signal satellites consistent with the actual GPS signal satellites at the real location? Thanks, expert!"

> 你好
> 
> 该代码实际上不是为欺骗而构建的，它更像是一个测试 GPS 接收器的模拟器。但是，此代码生成的基带信号可以通过适当的SDR设置进行传输，以欺骗GPS接收器。
> 
> 回到你的问题，它实际上需要一个RINEX文件，其中包含卫星的星历数据（你可以从NASA的网站获得它，也可以使用GPS接收器记录你自己的数据）
> 
> 它不需要实际坐标，但它需要一个目标位置来模拟该位置的可见卫星、它们的多普勒频移和它们的代码相位。
> 
> 不，它不使用受害设备上方的卫星信号（您想要欺骗）。该软件创建的每个通道中产生的卫星可能与直接出现在目标位置上的卫星不同。但是，它仍然完美地完成了它的工作。
> 
> 该软件只是根据为整个星座中的每颗卫星计算的仰角来猜测目标位置上的可用卫星。
> 
> 如果您想问其他任何事情，请随时。
> 
> 谢谢