osqzss/gps-sdr-sim #329: Moving receiver but Coordinates do not change

FayeWong98, 2022-06-22:
Why do I try to move the receiver after successful deception, but the coordinates do not change? Theoretically, the signal arrival time changes, and the pseudo range will also be different. The coordinates should be recalculated. I hope someone can answer for me. Thank you!

Imtiaz08, 2022-06-25:
Actually, the simulator generates a simulated signal with respect to a target spoofing location. You are not getting a real signal from satellites thus, your receiver will only be calculating the pseudorange from the spoofer not from the satellite thus it is not changing

FayeWong98, 2022-06-25:
> 

It is clear that all satellite signals are sent from one transmitter, but how to calculate the positioning of four satellites? All the signals come from the same transmitter, with the same pseudo range and time.And the movement does change the time when the deception signal arrives at the receiver, and the generated pseudo range should change. I am still a little confused.

Imtiaz08, 2022-06-26:
Technically, you will need to receive the signal from at least 4 satellites to calculate the pseudorange from all the satellites and then perform positioning. However, the simulator generates the satellite data with the given PRNs (Satellite ID) from the given RINEX files and then transmits each channel which appears to the receiver as if the signals are coming from 4 different satellites.

Imtiaz08, 2022-06-26:
Here's what happens in detail in the code
1. GPS Satellite Information is extracted from the RINEX file
2. Satellite position is calculated for the target coordinates using the ephemeris data and time
3. Pseudorange and elevation angles are calculated for each satellite 
4. Suitable satellites are selected based on their elevation angles (As 8 to 12 SVs can be visible at a time)
5. Each satellite and its data is kept in channels 
6. I/Q signal is generated from all the data and packed in a .bin file (because of the binary data)

IvanKor, 2022-06-26:
> Why do I try to move the receiver after successful deception, but the coordinates do not change?
for the same reason as
if you replace #define SPEED_OF_LIGHT 2.99792458e8 for example with #define SPEED_OF_LIGHT 4.0e8
the coordinates do not change

Imtiaz08, 2022-06-28:
The ephemeris time in the rinex file is basically the time at which the signal was received and stored in the Rinex file. 

Suppose you have recorded a signal and created a RINEX file yesterday it will have the time of yesterday. 

One more thing i.e. the time is in UTC which means I, a person from Pakistan, if reading the file, the time will be 5 hours early because we live in GMT+5

Imtiaz08, 2022-06-28:
Again, you are creating a scenario as if everything is constant and not moving (in the case of static signal mode). A receiver itself doesn't know what time is it and where is it going or coming from. 

When you transmit a spoofing signal it sends a uniform signal whose parameters begin from the time of transmission and is incremented gradually after every 30 seconds. 

In simple words, You are telling the receiver as if it is static and getting a signal from a satellite at a specific time and date

In the case of a moving scene, the receiver is told as if it is moving even if it is standing in one place. Here the code phases and doppler frequency of the signal is modeled in such a way that helps in the deception of the receiver

antex00715, 2023-01-03:
Imtiaz08 how to solve rinex file time delay