osqzss/gps-sdr-sim #330: Inconsistent signal USRP N210

Imtiaz08, 2022-06-30:
Hi I've been trying to spoof a Ublox C94 M8P receiver using USRP N210.

When transmitting the signals I can see satellites appearing in the u-center but it is for a very short time and only 1 or 2 svs are appearing. 

I have tried every solution mentioned in all opened and closed issues. None of them worked for me. 

I've tried transmitting the file from uhd.py  as well as tx_from_sample 

Can anyone please help me in this regard? 

I am working on my MS thesis and I really need this to work. Please

osqzss, 2022-06-30:
Which daughter board do you have in your USRP N210?

Imtiaz08, 2022-06-30:
Sbx v3

osqzss, 2022-06-30:
I will run a quick test with my N210+SBX. Which host OS do you use to transfer the I/Q samples?

Imtiaz08, 2022-06-30:
I am using Ubuntu sir

Imtiaz08, 2022-06-30:
I tried using sudo python3 gps-sdr-sim-uhd.py -t gpssim.bin -s 2500000 -x 0 and then increased it all the way up to - x 80

Also tried with the usr/lib/uhd/examples/tx_samples_from_file command mentioned in a few threads here

osqzss, 2022-06-30:
I need to know your gps-sdr-sim options, too.

osqzss, 2022-07-01:
@Imtiaz08 I have no issue to run a gps-sdr-sim scenario on my N210+WBX with the following steps:

$ ./gps-sdr-sim -e brdc2960.20n -l 35.681726,139.765120,10.0 -s 2500000
$ /usr/local/lib/uhd/examples/tx_samples_from_file --file gpssim.bin --type short --rate 2500000 --freq 1575420000 --gain 0

Again, you need proper RF attenuation between the N210 TX and the GPS receiver. I used 30dB+20dB=50dB attenuators in this test run. The host OS is Ubuntu 18.04.6 LTS, and UHD version is 4.2.0.0.

[image]

[image]

[image]

Imtiaz08, 2022-07-01:
Is there any way to receive the signal using gnss-sdr?

IvanKor, 2022-07-01:
gps-sdr-sim  ...
-----------------------------
Time into run = 300.0 [sec]
Done!.
Process time  = 59.1 [sec]
-----------------------------
300.0/59.1 = 5.076
need to use a laptop
providing 5 performance overlap
because it is a real time system

IvanKor, 2022-07-01:
however USRP N210 not the acc. device 
https://user-images.githubusercontent.com/7418842/176831950-3177afa1-32c6-4f95-be63-b81322d0050f.png
3D Acc.  [m] = 12.5 
error of 12.5 meters is too much for a signal without interference
my cheap hackrf gives a much smaller error
 [image]

Imtiaz08, 2022-07-01:
is there any way to read the signal using any software just to confirm the position extracted from the signal?

I mean by reading the signal using any software receiver such as RTKLib or GNSS SDRLib or something

Imtiaz08, 2022-07-01:
Thank you so much I have two questions:

1. Can I use the GNSS SDR Lib by Taroz Suzuki to read the bin file generated by the GPS SDR SIM?
2. What would be the file specifications such as dll threshold, pll threshold etc to read the file using GNSS-SDR-Lib

IvanKor, 2022-07-01:
> 1. Can I use the GNSS SDR Lib by Taroz Suzuki to read the bin file generated by the GPS SDR SIM?

Not

osqzss, 2022-07-01:
@Imtiaz08 https://www.youtube.com/watch?v=HfceySbuiBw

Imtiaz08, 2022-07-01:
> @Imtiaz08 https://www.youtube.com/watch?v=HfceySbuiBw

I am extremely sorry for being a noob but the configuration is the main issue. I've already watched this video but whenever I feed the file I am not able to track any signal. 

I know you must be very busy but if you could help me set a configuration file for the GNSS SDR or SoftGNSS MATLAB I will be very thankful to you for this act of kindness

osqzss, 2022-07-02:
@Imtiaz08 Have you even tried "My first position fix" tutorial for GNSS-SDR?
https://gnss-sdr.org/my-first-fix/

I have no problem to get navigation solutions from gps-sdr-sim I/Q samples by simply following their instructions:

[image]

[image]

[image]

[image]