osqzss/gps-sdr-sim #331: Spoofing iPhone device

sticky-tea, 2022-07-22:
Is it possible to spoof gps on iphone device? Works well with my android smartphone, but doesn't work for iphone (13 mini, ios 15.2). Tried with HackRF and LimeSDR. Tried to set air mode, turn off wifi and mobile data (there is no sim anyway), but no result. It either shows my current location, or it doesn't show any at all

Imtiaz08, 2022-07-25:
are you using the latest BRDC file?

sticky-tea, 2022-07-25:
> are you using the latest BRDC file?

Yeah, im using the latest BRDC. So is it even possible to spoof GPS in latest iphones with gps-sdr-sim? I just haven't found any info about it

Imtiaz08, 2022-07-25:
Have you tried doing the experiment indoors? Are there other satellite constellations as well? 

I think the latest smartphones are relying on multiple positioning i.e. IMU, Bluetooth, cell towers, etc that's why it is a bit difficult to spoof them. One of the methods that I tried to spoof my phone was turning off the auto-update time and date but since you aren't even using SIM card so I don't think this will be an issue

sticky-tea, 2022-07-25:
> Have you tried doing the experiment indoors? Are there other satellite constellations as well?
>
> I think the latest smartphones are relying on multiple positioning i.e. IMU, Bluetooth, cell towers, [...]

Yes, i do it indoors, wifi and bluetooth is off, no sim. It works fine with android device, doesn't work only with iphone

sticky-tea, 2022-08-08:
Well, now it works fine. To spoof iphone you should set up air mode and wait for some time (like one day), after that spoofing will work

closed 2022-08-08 by sticky-tea

vxling, 2023-04-19:
> 

I tried while SIM is enabled, with multi-sdr-gps-sim, use realtime mode.