osqzss/gps-sdr-sim #332: Finally! I spoofed my GPS receivers and low end android phones successfully but need help!

Imtiaz08, 2022-07-25:
Hello sir @osqzss, 

I don't have words to thank you for helping me so much and guiding me on different occasions. I have successfully spoofed my GNSS receiver and some low-end smartphones. I am studying the code and trying to learn it well so that I can work further and develop a receiver spoofer i.e. an algorithm that captures RAW signal, modify it with desired position and time and then retransmit it towards a victim device. 

I've learned a lot of things from this code. However, there are a few things that are still confusing me. 

1. what is the databit? why is only 1 databit being calculated from a dword of 1 satellite per time?
2. Where did the antenna pattern, sintable512, and costable512 values takem from?

I hope you will answer my questions whenever you have time. 

thank you once again and have a very nice day

With profound regards
Imtiaz nabi

Imtiaz08, 2023-05-09:
None yet, the admin seems too busy to answer

vxling, 2023-05-10:
Maybe you can share more details, eg: i replaced plutosdr crystal with 0.5ppm one, I got 3d fix with U-blox 8, with static mode, but doesn’t work on the iphone 14

Imtiaz08, 2023-05-25:
I can get a 3D fix with any ublox receiver using a HackRF one but this library seems to be not working on smartphones anymore. Maybe due to upgraded firmwares of the smartphone devices or the addition of other sensors.