osqzss/gps-sdr-sim #333: NMEA data instead of brdc file

ajayavgarde, 2022-08-18:
Can the NMEA data directly be passed from the GPS receiver to the program instead of brdc file

osqzss, 2022-08-18:
NMEA format doesn't contain satellite ephemeris data.

ajayavgarde, 2022-08-18:
is there a way that i can extract the satellite ephemeris data using GPS receiver

ajayavgarde, 2022-08-18:
can I get the ephemeris data using ublox instead of downloading it from the website?

osqzss, 2022-08-18:
You have to extract the ephemeris by yourself from UBX-RXM-SFRBX:
https://www.u-blox.com/en/technologies/gnss-raw-data

ajayavgarde, 2022-08-18:
Okay thanks, I will convert the data from UBX-RXM-SFRBX to RINEX format and pass it to gpssim . will it work?

osqzss, 2022-08-18:
You can use RTKCONV from RTKLIB to generate RINEX navigation file from UBX-RXM-SFRBX.
https://github.com/tomojitakasu/RTKLIB

ajayavgarde, 2022-08-19:
Is there a dll file of this program

ajayavgarde, 2022-08-19:
okay, thank you for the information, and nice work on the code!

ajayavgarde, 2022-08-22:
Should i keep on download the brdc file every time they upload on NASA website

Imtiaz08, 2022-09-05:
You need the latest satellite ephemeris information. So technically, yes you will need to download the latest file unless and until you are using your own data.