osqzss/gps-sdr-sim #334: SyntaxError & AttributeError in gps-sdr-sim-uhd.py

aeaotst, 2022-08-19:
I've been working on getting a signal out of a USRP B200 mini-i for the last couple days and I think I'm almost there but I've hit a bit of a roadblock. I have successfully generated `gpssim.bin` using the provided broadcast data and the `satellite.csv` file. I did so using the pre-packaged .exe. I've installed GNURadio and the UHD packages using radioconda on Windows 10. I imported the `gps-sdr-sim-uhd.py` file to my working directory and ran the following command from the Conda Prompt packaged with GNURadio:

`C:\...\gps-sdr-sim\python gps-sdr-sim-uhd.py -t gpssim.bin -s 2500000 -x -40`

Initially I had a SyntaxError on line 84 of the script caused by the expression `raise SystemExit, 1`. Removing the `, 1` from the line easily fixed that issue, though what impact this may have is unclear to me.

Now, running the same command, I get:

`Traceback (most recent call last):
  File "C:\...\gps-sdr-sim\gps-sdr-sim-uhd.py", line 91, in <module>
    tb = top_block(options)
  File "C:\...\gps-sdr-sim-uhd.py", line 24, in __init__
    self.uhd_usrp_sink = uhd.usrp_sink(options.args, uhd.io_type_t.COMPLEX_FLOAT32, 1)
AttributeError: module 'gnuradio.uhd' has no attribute 'io_type_t'`

After digging through documentation, I've come to the conclusion that `io_type_t` has been deprecated in the gnuradio libraries. I've found references to it in usrp_sink_impl.cc releases dating back to 2019, but nothing recent. Since I'm using the most recent version of GNURadio this is likely the source of this problem, though I've been having difficulty building older versions so I haven't yet tested it.

I'm also noticing that my IDE is highlighting the expression `raw_input('Press Enter to quit: )`. As far as I can tell there is no method defined as `raw_input()` so I expect this will be the next issue I face.

osqzss, 2022-09-08:
Try tx_samples_from_file:
https://github.com/osqzss/gps-sdr-sim/issues/330

sonohikari, 2023-12-28:
Hello, I have encountered the same problem as you. May I ask how you resolved it in the end

aeaotst, 2024-01-05:
Ultimately I didn't find a direct solution. I bypassed the problem by running the GNURadio GUI and creating a pipeline that matches the structure of the Python code.

sonohikari, 2024-01-08:
Sincerely thank you for your reply.
Ultimately I also bypassed this problem by trying the tx_samples_from_file mentioned by the author, and it worked for my problem.
Thank you for your reply again.