osqzss/gps-sdr-sim #337: Running GPS-SDR-SIM without adding noise

fener1efes2, 2022-08-28:
GPS-SDR-SIM adds noise in band purposefully.
I want to observe its BPSK signal.
How can I turn off adding noise property?

[image]

osqzss, 2022-08-28:
gps-sdr-sim generates pure baseband signals. No noise is added.

fener1efes2, 2022-08-28:
> gps-sdr-sim generates pure baseband signals. No noise is added.

I shared output of the HackRF transmitting signal produced by the GPS-SDR-SIM. 
As you can see, the waveform is not as it is expected.
I would expect a waveform as shown below.
 
[image]

osqzss, 2022-08-28:
Yes, that's exactly what you can expect to see. It seems that the HackRF did not transmit the baseband samples properly. You have to set the I/Q sample format to 8-bit with -b option for the HackRF.

aeaotst, 2022-08-29:
I get similar results using the USRP B200 mini-i regardless of whether I use 16-bit or 8-bit. My receiver is reading the signal and getting a fix so perhaps not an issue, though I have to wonder if there's any implication for C/N.

osqzss, 2022-08-29:
Here's a sample PSD plot of the L1 C/A signal with a 2.6MHz sampling rate and a 2.5MHz double-side bandwidth. You can see the main lobe on center and partial side lobes on both ends.

[image]

fener1efes2, 2022-08-30:
why is it so noisy? I tested another commercial GPS simulator and it gives an output as shown below. 

[image]

fener1efes2, 2022-09-03:
> Here's a sample PSD plot of the L1 C/A signal with a 2.6MHz sampling rate and a 2.5MHz double-side bandwidth. You can see the main lobe on center and partial side lobes on both ends.
> 
> [image]
@osqzss 
Is the bandwidth of the signal 3.6MHz? If I see the span correctly which is 6MHz, then the span must be 3.6MHz?

Imtiaz08, 2022-09-05:
I think the transmitter and the receiver component added noise can't be neglected. Have you checked probing the signal in the software without transmitting it?

IvanKor, 2022-09-05:
> I think  ...

GPS uses pseudo-noise 1023-bit  technology.

By connecting the output RF of the HackRF transmitter [Real-time GPS signal simulator based on GPS-SDR-SIM](https://github.com/IvanKor/hrfGPS) via an attenuator to the NEO-M8T antenna input for 1.5 hours and turning on all corrections Solution: SINGLE [RTKLIB](http://www.rtklib.com) make sure that during this time the total error (RMS) ~ 0.1 meters.

[image]

If use USRP B200 (16-bit) instead of HackRF (8-bit), then instead of RMS~0.1m get RMS<0.01m

**If your simulator has RMS>0.1m (1m or more) then this simulator is not correct.**

closed 2022-09-25 by fener1efes2