osqzss/gps-sdr-sim #340: How do I buildthe plutoplayer for windows 11?

William-Proctor, 2022-09-08:
I have tried to do the same method to build the pluto player as the gps-sdr-sim e.g opening a c++ console project in visual studios and adding the plutoplayer.c, getopt.c , & ad9361_baseband_audio_rate.c .

I have tried with and without ad9361_baseband_audio_rate.c but in both cases I get the same results a lot of errors and a failed build here are the build errors for with the ad9361 file.

1 - 16:
[image]
16 - 31 :
[image]

And here are the 26 errors without the ab9361 file.
[image]

Am I missing a file that contains the imp_iio content or should I be building this in some other way?

I'm running windows 11 and using a adalm-pluto rev.c which is on V0.35 connected by a usb and it has worked for listening so far.
Any assistance would be much appreciated, thanks.

osqzss, 2022-09-08:
You need libiio.
https://github.com/analogdevicesinc/libiio

William-Proctor, 2022-09-08:
Thanks for getting back to me so quickly but Ive installed it using the libiio-0.24.gc4498c2-Windows-setup.exe from https://github.com/analogdevicesinc/libiio/releases/download/v0.24/libiio-0.24.gc4498c2-Windows-setup.exe link should I have done it a different way if so how?

osqzss, 2022-09-08:
Download Windows-VS-2019-x64.zip or Windows-VS-2022-x64.zip. You can find libiio.lib that you need to link to.

William-Proctor, 2022-09-08:
Thank you so much it has built successfully .

closed 2022-09-08 by William-Proctor

Rupak910, 2024-01-15:
Hello sir, Can you please explain how you linked the libiio.lib library. It will be a great help.

Rupak910, 2024-01-15:
> Download Windows-VS-2019-x64.zip or Windows-VS-2022-x64.zip. You can find libiio.lib that you need to link to.

> Thank you so much it has built successfully .

hello sir, can you please help me with how you linked the libiio.lib file and where.