osqzss/gps-sdr-sim #341: fake doppler

kamitanihidetoshi, 2022-09-09:
Hi!

sorry if my English is bad.
Is it possible to add a fake Doppler to the simulation to make it more realistic?

Imtiaz08, 2022-09-09:
You are actually manipulating doppler for specific locations. What exactly do you mean by fake doppler?

kamitanihidetoshi, 2022-09-15:
If you want to send a false signal, the transmitter is fixed. I don't think we can create a signal that looks like a real satellite that is actually moving. Therefore, we want to generate a false signal as if a fixed false signal transmitter were moving.
In other words, we think that false Doppler is a mechanism to make a fixed transmitter appear to be moving.

Sorry if I was confused.

Imtiaz08, 2022-09-19:
Yes, it is definitely possible to add fake doppler the way that you've described. You will need to use advanced techniques such as the Kalman filter or machine learning models to predict the previous states and estimate the new stats. 

It isn't easy to do but it is definitely possible I think

IvanKor, 2022-09-19:
> It isn't easy to do but it is definitely possible I think

Doppler shift
https://github.com/osqzss/gps-sdr-sim/issues/54

closed 2022-10-12 by kamitanihidetoshi