osqzss/gps-sdr-sim #342: Problems geting 3D fix when using dynamic mode

wthiam002, 2022-09-13:
I am having trouble getting a 3D fix on u-center when using my own user motion file in lat, lon, height format. 
However I have no issues getting a fix when testing using the circle_llh.csv present in the source folder.
Are there any requirements when creating your own user motion file that I am missing such as minimum time required etc. ?
I am also using the latest brdc file as of today but could not get the fix.

Any help would be appreciated, thank you.

ark-1313, 2022-09-14:
I think it is probably the oscillator source. Which SDR are u using ?

wthiam002, 2022-09-14:
> 

I'm currently using a HackRF One

ark-1313, 2022-09-15:
Ok, try using a better oscillator source instead of driving from native oscillator on it.

ark-1313, 2023-05-11:
Did you try using a better oscillator source ? Also,  did u use satgen software to create your own user motion file ? Are you able to get a fix for static co-ordinates ?

vxling, 2023-05-11:
I cant get a stable 3d fixed with static mode.

for pluto SDR i replaced the crystal with a 0.5ppm model: DSB321SDN 40M 40.000MHZ 3225 TCXO XTW40000PBA.

Receiver is U-Blox M8030 72ch.

but my iphone 8 and iphone 14 can’t be spoofed. The phone with SIM and 4g turned on.

for dynamic mode, I didn’t test yet

ark-1313, 2023-05-12:
Ok, I understand that you are trying to get a static fix. I too have tried with Adalm, but didn't get much success. Then I switched to BladeRF & also disciplined the clock using an external Oscillator : http://www.leobodnar.com/shop/index.php?main_page=product_info&cPath=107&products_id=301&zenid=00f38f6a4772c583eec624ecc6cbb773

vxling, 2023-05-12:
Thank you very much for your information sharing.
Now the reason is clearly because of the accuracy of the oscillator, maybe use an external vctxco or ocxo just like https://www.mictronics.de/posts/ADLAM-Pluto-OCXO-Mod/, with a higher accuracy 0.01ppm. But I don’t test it yet.

my idea is, can we add some software filter to I/Q to improve the output signal quality with current 0.5 ppm oscillator?

ark-1313, 2023-05-13:
Sorry, I wasn't able to access the link that you have provided. Yet, like you have said, the accuracy is better, so you should give it a try. But I am not sure of the RF pad & track for the oscillator in Pluto i.e., if you provide signal from an external oscillator , I am not sure of the attenuation of the signal. However, in boards like USRP, there is an external connector for the clock which can maintain the impedance.

I think the signal generation from the transmitter is good enough, as it depends upon the RF front end which is an AD9363/64 on the Pluto and therefore we only need to concentrate upon the input clock. However, I might be wrong on this & it will be nice in case some experts comment upon this.

osqzss, 2023-05-13:
@wthiam002 Regarding your motion file issue, this may help: https://github.com/osqzss/gps-sdr-sim/issues/301

vxling, 2023-05-18:
> 

That means you locked the frequency with an external clock source. Then signal quality improved and worked with the receiver devices. I will try to do that. Thank you.