osqzss/gps-sdr-sim #343: BRDC/Hourly data

ark-1313, 2022-09-14:
Hi,

We are using the CDDIS hourly/daily data for the navigation data generation. We wish to generate the data for the current UTC. However, from the downloaded data it says that the timing (TOC & TOE) are invalid and hence will not be able to generate for the required time. This requires us to wait until the new data is released from the cddis. If we try over-riding using the 'T' option, it is observed that the  receiver does not spoof at all.
1. Is there any other way through which the TOC/TOE can be modified to get the data for the current UTC from the hourly/brdc file ?
2. We came across another issue in which you suggested to use RTKLib with a receiver to process the raw data in order to obtain the ephemeridis. How long will the validity of a data obtained in such a manner be & how long should such a recording be done?

With Regards,
ARK.

ark-1313, 2022-09-15:
Any help, anyone ?

osqzss, 2022-09-15:
2. It requires only a 18-sec long navigation massage (subframe 1-3), which repeats every 30 seconds.

ark-1313, 2022-09-15:
& we can generate data through the tool for the next 1 hour ?

What about (1) ?

osqzss, 2022-09-15:
A broadcast ephemeris is typically valid for about two hours.
Regarding question 1, the answer is no.

ark-1313, 2022-09-15:
Thanks a lot. Wonderful tool BTW !!

closed 2022-09-15 by ark-1313