osqzss/gps-sdr-sim #346: Why phase_ini = (2.0 * r_ref - r_xyz ) / LAMBDA_L1 ?

shuwang1, 2022-09-30:
Hey @osqzss and all GNSS experts,

I guess this may not be a real issue but my personal curiosity.  Why is the initial phase initialized in such a way?  

TIA,

Shu

Imtiaz08, 2022-11-10:
@shuwang1 so how should it be? Like I also have an answer to this question but I would like to know your input first so that I can understand what you actually want to say. I mean do you think there is something wrong with the phase initialization? 

I think this is the initial signal phase that is calculated between the user position and the broadcast orbit i.e Medium Earth Orbit (MEO) since all GPS satellites are in MEO. Once we get the initial phase only then we can do alteration via doppler shift.

cjb-eng, 2022-12-29:
I actually had the same question as @shuwang1. Working with truth observation files from other simulators, the phase is typically initialized as the number of carrier cycles associated with the pseudorange calculated from the ephemeris. In the context of the code, that would translate to phase_ini = r_xyz/LAMBDA_L1. It seems the current initialization introduces a large number of cycle ambiguities at the start of the simulation.

osqzss, 2022-12-29:
Please ignore that line. It is still experimental.