osqzss/gps-sdr-sim #348: Broadcasting the gpssim.bin from the pluto via a usb not working

William-Proctor, 2022-10-10:
As above I'm trying to broadcast from a Pluto using a runme0.sh script the script was pulled from https://rntfnd.org/wp-content/uploads/Regulus-PPT-to-PNT-AB.pdf and consists of three lines of code.

iio_attr -a -c ad9361-phy TX_LO frequency 1575420000
iio_attr -a -c -o ad9361-phy voltage0 sampling_frequency 2600000
cat /media/sda1/spoof.bin | iio_writedev -a -b 50000 cf-ad9361-dds-core-lpc

So I renamed my gpssim.bin output to spoof.bin while generating it using the -o command an put it and the runme0.sh on a usb formatted to FAT32.
I then plugged this in, and the PlutoSDR lit up on LED1 as well as the ready light but it was not outputting correctly I then realized I was on a windows system and needed to add LF to each line so that the Pluto would run the next line however this has not fixed the output am I missing something obvious?

My pluto rev.C is using a external clock at 40mhz that works when I output from the plutoplayer  from the laptop but not running from the USB I'm hoping its something simple like the micro USB to USB adapter they sent is not a USB_On-The-Go so I have ordered one online however I still think there will be something else wrong either with the code above or my understanding of linux systems. 

Any assistance would be appreciated as SDR is pretty new to me I've been learning it for the last 2 months.
Thank you in advance.

Imtiaz08, 2022-11-10:
@William-Proctor Would you like to show us some screenshots of the problem that you are currently facing? I mean is there any kind of error that you are getting or you are not receiving the signal.

William-Proctor, 2022-11-10:
I'm not getting any error messages. I'm just not receiving the signal and when I added a reboot on the next line so that I would have some indication of if the pluto was going through the code and transmitting the Bin file it reboots really quickly so I added some touch commands to files that didn't exist to see how far the pluto goes through the code and it generated files after each line so I know it sees the command it just doesn't transmit.

Here is the test code:

iio_attr  -c ad9361-phy TX_LO frequency 1575420000 LF
touch /media/sda1/0.txt LF
iio_attr  -c -o ad9361-phy voltage0 sampling_frequency 2600000 LF
touch /media/sda1/1.txt LF
cat /media/sda1/spoof.bin | iio_writedev  -b 50000 cf-ad9361-dds-core-lpc LF
touch /media/sda1/2.txt LF
reboot LF

I don't know much about linux on the pluto or the OTG so I thought I might be calling these functions incorrectly or was missing something obvious.
Thanks for responding and thanks in advance for any insight and help.

osqzss, 2022-11-25:
Try pluto_player: https://github.com/osqzss/gps-sdr-sim/tree/master/player

William-Proctor, 2022-11-25:
Yes that works brilliantly, however I was trying to use the pluto's otg to see how the player functions like which steps do what on the pluto, as far as I understand it the pluto player generates a small buffer and sends packets to the pluto which transmits on the GPS frequency at a sample rate of 2600000 in the I/Q format from a bin file.

So far I roughly understand the generation and transmitting using the pluto_player.

However, what I was trying to do was see if using GPS sim to generate the bin and the pluto's on the go runme files could work to create a semi-portable GPS spoof device.

Hence me sharing the code below as my start point.
`iio_attr -a -c ad9361-phy TX_LO frequency 1575420000
iio_attr -a -c -o ad9361-phy voltage0 sampling_frequency 2600000
cat /media/sda1/spoof.bin | iio_writedev -a -b 50000 cf-ad9361-dds-core-lpc`
 
 any ideas would be appreciated.

osqzss, 2022-11-25:
The runme.sh is just a shell script. Why don't you build pluto_player for PlutoSDR standalone and call it from runme.sh?
https://wiki.analog.com/university/tools/pluto/devs/embedded_code

Regarding your IIO issue, I think you could get better support from ADI EngineerZone:
https://ez.analog.com/adieducation/university-program/