osqzss/gps-sdr-sim #349: Error "exited with code 1" in visual studio

HaXlezi, 2022-10-11:
Hello guys, I went through the manual how to start application in Visual Studio. I imported nescesary files but after launching the code at the end of console app the text "The program ... has exited with code 1"  appears. I am just begginer in C++ programming so the problem could be very simple. Can you help me please? I attach sreenshot after debugging

[image]

Imtiaz08, 2022-11-10:
I am not quite sure what you are actually doing out there. But from what I understand you might be looking for a way to debug the code and study it line by line right? 

If that's true then you will need to make some changes to the code because it takes user inputs at some points. You will need to change those lines so that the code takes some default values and return the process results. If you need any help in this regard I will be more than happy to help. 

:) I am been there and I truly understand so don't worry about asking questions and learning

William-Proctor, 2022-11-14:
Right if you want to run it as an application you have to Select "Release" in Solution Configurations drop-down list & Build the solution. after that go through the bin release folder in  "ProjectName"\x64\Release  and open it using CMD via PathToExe/program.exe   it should then give you some possible inputs so you then repeat the command and add your inputs.
e.g. PathToExe/program.exe  -l 123 , 1341 , 10

HaXlezi, 2022-11-16:
> Right if you want to run it as an application you have to Select "Release" in Solution Configurations drop-down list & Build the solution. after that go through the bin release folder in [...]

Oh god, that was way simpler than I was thinking. Yeah now it's working right. Thank you so much

closed 2022-11-16 by HaXlezi