osqzss/gps-sdr-sim #353: Generate QZSS

kurokkasu, 2022-10-31:
I am making a QZSS signal.
The signal does not become the green used for positioning.
Please advise what should be changed.
[image]

osqzss, 2022-11-02:
It turns green when it is used in navigation. You never get valid 3D navigation fix with only three satellites.

kurokkasu, 2022-11-06:
How do I know if it's the correct signal?

osqzss, 2022-11-06:
Add GPS satellites.

kurokkasu, 2022-11-06:
If gps is added and qzss turns green, does that mean it's okay?

osqzss, 2022-11-06:
Plus you have to get proper navigation solutions.

closed 2022-11-06 by kurokkasu

osqzss, 2022-11-06:
Would you please send a pull request to add your QZSS signal generation code to the gps-sdr-sim? Your contribution would be greatly appreciated.

kurokkasu, 2022-11-10:
[image]
I tried receiving again.
Is this a success?

IvanKor, 2022-11-10:
too bad values Residual