osqzss/gps-sdr-sim #354: Please tell me how to display the map

kamitanihidetoshi, 2022-11-01:
[image]
I don't know how to display the map. I installed Google Earth.

Imtiaz08, 2022-11-10:
Aren't you on the wrong thread? I think this is related to GNSS-SDR and GPS SDR SIM is something different. The former is a receiver while the latter is a signal generator.

kamitanihidetoshi, 2022-11-21:
Sorry if my understanding is wrong. This screen is GNSS-SDR. What settings should I do to display the map?

kamitanihidetoshi, 2022-11-23:
I don't understand about Esri contributors.

osqzss, 2022-11-23:
Wrong place to ask. Pop the question to gnss-sdr-monitor guys. 
https://github.com/acebrianjuan/gnss-sdr-monitor

closed 2022-11-23 by kamitanihidetoshi