osqzss/gps-sdr-sim #356: Receiving only one satellite signal on Android via HackRF

HaXlezi, 2022-11-21:
Hello guys,
I was trying to spoof location on Samsung Android device using HackRF One + Portapack. I think I generated file correctly using manual on [https://github.com/eried/portapack-mayhem/wiki/GPS-Sim](url). After starting the simulation my phone was receiving signal from **only one satellite** which is not enough to get a fix obviously (shown on picture). It is weird because it looks like the SDR transmit some signal but only for one satellite. Changing of transmitting power didn't change anything. I tested 3 Android phones. Airplane mode was on.
Do you have any ideas?
Thanks a lot!

[image]

osqzss, 2022-11-25:
You might need a better clock: https://github.com/osqzss/gps-sdr-sim/issues/290

liwuxin, 2022-12-01:
I have the same problem as you. The reason is that the satellite calendar of mobile phone gps is different from the simulated satellite track. Solution:  a.To reset the cached satellite data in the GPS, you can try to restart the mobile phone in flight mode, and do not connect to the Internet during the process. b.Satellite calendar with the closest download time

HaXlezi, 2022-12-01:
> I have the same problem as you. The reason is that the satellite calendar of mobile phone gps is different from the simulated satellite track. Solution: a.To reset the cached satellite data in the [...]

Oh god, this one really helps, thank you! I knew about the airplane mode but I didn't try restarting. This was really heplful. Thanks man!

HaXlezi, 2022-12-01:
I'm in London now! (In fact I'm in Prague ;) ) Thx
[image]

Commander-Crash, 2023-03-29:
> I have the same problem as you. The reason is that the satellite calendar of mobile phone gps is different from the simulated satellite track. Solution: a.To reset the cached satellite data in the [...]

TYSM