osqzss/gps-sdr-sim #358: MAX_CHAN value

fener1efes2, 2022-12-29:
What does MAX_CHAN value define?
Which definition below is correct?

1-) Exact number of satellites that is being simulated.
For example, if I set MAX_CHAN to 5, the code will simulate 5 satellites all the time.

2-) Maximum number satellites that is being simulated.
If I set, MAX_CHAN to 5, the code will simulate up to 5 number of satellites.
Sometimes number of simulated satellites will be 4, sometimes 5.. etc..

osqzss, 2022-12-29:
No. 2.

fener1efes2, 2022-12-30:
Can you give a little background?
Why does the simulation act like that, why does not always simulate 5 channels?

osqzss, 2022-12-31:
The number of visible satellites could be less than MAX_CHAN.

fener1efes2, 2022-12-31:
I set MAX_CHAN=1
But the GPS receiver observed more than one satellite. Why does that happen?

closed 2023-01-04 by fener1efes2

Imtiaz08, 2023-05-25:
yeah I noticed some strange because I was also getting BDS satellites while receiving the signal using ublox zed F9p and hackRF one. I was feeding the signal directly via sma to sma 

maybe because of the C/A code errors IDK but I forgot to take a screenshot