osqzss/gps-sdr-sim #361: No signal received on bladeRF 2.0

emeraldMaster, 2023-01-08:
Hi all! I'm trying this software with bladeRF micro a4, and got a problem: SDR transmits _something_, but it has no effect on my other devices. Here is what I tried:
1. downloaded latest `brdc` file from https://cddis.nasa.gov/archive/gnss/data/daily/2023/brdc/
2. generated bin file: `./gps-sdr-sim -e ./brdc0080.23n  -l 30.286502,120.032669,100`
```Using static location mode.
xyz =  -2758918.6,   4772301.1,   3197889.4
llh =   30.286502,  120.032669,       100.0
Start time = 2023/01/08,00:00:00 (2244:0)
Duration = 300.0 [sec]
02  178.2   6.5  24710954.7  16.7
10  323.7  47.4  21614380.7   6.1
12  124.0  32.1  22549840.6   9.2
15   76.7  23.0  23331908.4  10.9
18  205.9  33.1  22448819.9   8.4
22  272.1   4.5  25468927.2  11.2
23    8.5  82.2  20179718.5   5.0
24   39.6  52.7  20875186.5   6.0
25  157.8  25.6  23114875.6  10.5
31  223.2   0.5  25582049.3  16.2
32  290.6  27.5  23184214.9   8.1
Time into run = 300.0
Done!
Process time = 26.2 [sec]
```
3. updated bladeRF script to make it compatible with 2.0
4. tried transmitting both via `bladeRF-cli` and `bladeplayer`, output is 
```
Opening and initializing device...
TX frequency: 1575420000 Hz
TX sample rate: 2600000 sps
TX bandwidth: 2500000 Hz
TX VGA1 gain: -25 dB
Running...
```

It seems to be running, but all GPS devices on 1575,42 just ignore SDR output. Any suggestion on what I'm doing wrong?

osqzss, 2023-01-08:
This might help: https://nuand.com/forums/viewtopic.php?t=4984

czbxzm, 2023-09-05:
hi have you fixed this problem,caused by bladerf2.0?