osqzss/gps-sdr-sim #363: Real time GPS Simulation

shauryaneel, 2023-01-11:
Hello All,

I am trying to simulate the GPS real time. I get the GPS data (lat, long, elevation) from Carla simulator (driving simulator). The GPS data gets updated every 100ms in carla simulator and I want to send these dynamic GPS data to HackRF to perform gps spoofing. 

I understood that dynamic gps spoofing can be done by using the static csv files available in this project and this csv file should be given input. In my scenario I directly want to send the gps coordinates every 100-200ms to hackrf. Is it possible by using this Software package ?

osqzss, 2023-01-12:
GPS-SDR-SIM is not intended to run in real time. For HackRF, try multi-sdr-gps-sim:
https://github.com/Mictronics/multi-sdr-gps-sim

shauryaneel, 2023-01-13:
Thanks a lot for your quick reply, I will try doing with the https://github.com/Mictronics/multi-sdr-gps-sim.

Gerald0217, 2023-11-07:
Hello, respected experts，can I directly input latitude and longitude to transmit GPS simulated signals without generating the gpssim.bin file?

jvictors-tp, 2023-11-27:
@Gerald0217 with the tool linked above, yes. 

Note that you will absolutely need to use a Faraday cage for this either way.