osqzss/gps-sdr-sim #364: Can't get a stable 3D fix with Adalm-Pluto

Papabearer, 2023-01-12:
Hello everybody, 

At the moment I am having an issue regarding getting a stable 3D gps fix from my ADALM-Pluto and would appreciate some recommendations on how to get a more consistent fix. Just as a heads up, I'm using the linux terminal with Debian.

This is what I compiled:
> ./gps-sdr-sim -e brdc0060.23n -l 30.286502,120.032669,100

What I use to run it:
> player/plutoplayer -t gpssim.bin 

So far I have tried adjusting the bandwidth and attenuation to no avail. These are some images I captured regarding what my gps-receiver finds when I run the script. 

[image]

So my question is, is there anything else I can try to maintain a stable 3D fix? Help would be greatly appreciated 😄

osqzss, 2023-01-12:
Try a good exernal clock. The frequency offset of the onboard XO could be too big.

vxling, 2023-05-08:
i cant get a stable 3d fixed too, for pluto SDR i replaced the crystal with a 0.5ppm model: DSB321SDN 40M 40.000MHZ 3225 TCXO XTW40000PBA, I don’t know why.

Receiver is U-Blox M8030 72ch.

i also tested with hackrf One with an external clock, no 3d fixed too.

yosefzh, 2024-07-29:
> I got 3d fix with ublox now, but the iphone can’t locate

i have the same issue i managed to spoof my android phone but couldn't spoof my iphone, do you have any idea why ?