osqzss/gps-sdr-sim #368: Minimal hardware for GPS simulation

denisjoshua, 2023-03-27:
Hi there, 
I’m a beginner in the world of radio so I want first to ask is possible to do what I want… then I’ll study how to do it.

What I want is to simulate a GPS position of my home,
in order to use my mobile phone to login on my office work (I work remotely).
Like this, I can login at work office when I’m not exactly on my home position.

Please indicate me if you want, the possibility to do this with a minimal hardware cost.
I don’t need to do other radio thinks… I don’t need to receive other RF freq.
I just need to simulate the GPS for about 20 seconds until I login in my work office remotely.

Have you a link that explain me step-by-step how to simulate GPS using minimal hardware please ? 

Thanks a lot

osqzss, 2023-03-27:
This may help: https://drfone.wondershare.com/fake-location/gps-spoofing-with-hackrf-from-windows.html

denisjoshua, 2023-03-27:
Thanks a lot for the link... I'll study it. 
Now I just search the model of HackRF to buy :-) 

Thanks again 
Denis

closed 2023-07-07 by osqzss