osqzss/gps-sdr-sim #369: Transmit data with PlutoSDR but no signal

John132-LI, 2023-04-11:
I follow the instructions to build the project on Ubuntu 20.04. Then create "gpssim.bin" and try to transmit to Pluto SDR via "plutoplayer".
Command lines are shown as:
`./gps-sdr-sim -e brdc0360.23n -l 1.378898,103.807031,100 -d 300 -s 4000000`
`./plutoplayer -t gpssim.bin -a -80.0 -b 5.0`

Everything seems fine and the console shows "Transmit starts"
but when I use "GPS test" to check the sinal , no satellite is found(Andorid,only on Plane mode)
I don't know if the device is transmitting or not. Anyway to trace it?
How could I find out If something goes wrong ?
Any help would be sincerely appreciated :)

vxling, 2023-04-19:
I checked with UBlox, it received signals but the data invalid.

vxling, 2023-04-19:
Have you replaced the TXCO on Pluto?

osqzss, 2023-04-19:
@John132-LI  The original XO on the PlutoSDR board has a very large frequency offset. You need to supply more stable external clock input or replace it to a TCXO.

John132-LI, 2023-04-19:
Thanks for your help. 👍 
It turns out that the TX1 channel goes wrong.
I get rid of it and finnally receive signal.
But when I try another test, this error occurs:

`* Transmit starts...
ERROR: READ LINE: -32
Error pushing buf -32
Done.
ERROR: READ LINE: -32
ERROR: READ LINE: -32
`
What's happening?

DanishAliChanna, 2023-11-20:
hello John132-LI

how you solved your problem? i am facing same issue as yours? it appears transmit starts but then no output has shown.

John132-LI, 2023-11-20:
> hello John132-LI
> 
> how you solved your problem? i am facing same issue as yours? it appears transmit starts but then no output has shown.

@DanishAliChanna  Hi,thanks for you reply.
My problem is that there was a wrecked capacitor on the motherboard. It took me a few days to find that.
So I replaced it with a new one then everything went fine!
Unfortunately it's been more than half a year and I've forgotten the details.
But I assure that using "Player.c" is the easiest way to create the ouput signal.
May all go well with you！

DanishAliChanna, 2023-11-20:
Hello dear,
Thank you for your reply. actually i want to ask you, have you modified
TCXO in plutoSDR for transmitting gps signals? i have read the thread that
the owner was saying to update this. I am stuck with this problem. can you
please help me out to get rid of this issue. since last month i am
struggling with this. can you send me the default setup of your project? or
can you give me your whatsapp or any other kind of contact so i can
discuss with you on this topic. to be very honest i am not going to take
your long time i just want your 5 minutes to discuss the default setup of
this project.

thank you once again for your kind response, i am waiting to hear from you.

regards

Rupak910, 2024-01-15:
> I follow the instructions to build the project on Ubuntu 20.04. Then create "gpssim.bin" and try to transmit to Pluto SDR via "plutoplayer". Command lines are shown as: `./gps-sdr-sim -e [...]

hello john132-LI

i am also using pluto sdr.
But i cant transmit the signal. It is saying plutoplayer is not recognised. As you have already used it, can you help me out with what i should do here.