osqzss/gps-sdr-sim #375: transmitting

YY-chen555, 2023-06-28:
When I type "python3 gps-sdr-sim-uhd.py -t gpssim.bin -s 2500000 -x 0", nothing happens and my device is detectable by the virtual machine.
[image]

osqzss, 2023-06-29:
Try tx_samples_from_file instead.

YY-chen555, 2023-06-30:
> 请尝试 tx_samples_from_file 代替。
Yes, I used that command and got the following result, I am not sure if this means that USRP is sending the signal. Because on the PCB the LED on the side of the TX is on, but I get a failure when I use the GPS-test software on my phone to check if it is virtually positioned.

`tx_samples_from_file --file gpssim.bin --type short --rate 2500000 --freq 1575420000 --gain 0

Creating the usrp device with: ...
[INFO] [UHD] linux; GNU C++ version 11.2.0; Boost_107400; UHD_4.1.0.5-3

RtApiPulse::DeviceInfo pa_context_connect() failed: Connection refused

ALSA lib confmisc.c:855:(parse_card) cannot find card '0'
ALSA lib conf.c:5178:(_snd_config_evaluate) function snd_func_card_inum returned error: No such file or directory
ALSA lib confmisc.c:422:(snd_func_concat) error evaluating strings
ALSA lib conf.c:5178:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1334:(snd_func_refer) error evaluating name
ALSA lib conf.c:5178:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5701:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib control.c:1528:(snd_ctl_open_noupdate) Invalid CTL default
ALSA lib confmisc.c:855:(parse_card) cannot find card '0'
ALSA lib conf.c:5178:(_snd_config_evaluate) function snd_func_card_inum returned error: No such file or directory
ALSA lib confmisc.c:422:(snd_func_concat) error evaluating strings
ALSA lib conf.c:5178:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1334:(snd_func_refer) error evaluating name
ALSA lib conf.c:5178:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5701:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib control.c:1528:(snd_ctl_open_noupdate) Invalid CTL default
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
Using Device: Single USRP:
  Device: B-Series Device
  Mboard 0: B210
  RX Channel: 0
    RX DSP: 0
    RX Dboard: A
    RX Subdev: FE-RX2
  RX Channel: 1
    RX DSP: 1
    RX Dboard: A
    RX Subdev: FE-RX1
  TX Channel: 0
    TX DSP: 0
    TX Dboard: A
    TX Subdev: FE-TX2
  TX Channel: 1
    TX DSP: 1
    TX Dboard: A
    TX Subdev: FE-TX1

Setting TX Rate: 2.500000 Msps...
[INFO] [B200] Asking for clock rate 40.000000 MHz... 
[INFO] [B200] Actually got clock rate 40.000000 MHz.
Actual TX Rate: 2.500000 Msps...

Setting TX Freq: 1575.420000 MHz...
Setting TX LO Offset: 0.000000 MHz...
Actual TX Freq: 1575.420000 MHz...

Setting TX Gain: 0.000000 dB...
Actual TX Gain: 0.000000 dB...

Checking TX: LO: locked ...
`

aprime2t, 2023-07-04:
Hello, I've been trying to download the gps-sdr-sim.zip file but it's missing from the release link. Could someone please send me the file? I've tried everywhere to try to find the file but with no luck!

My email is aprime2t@gmail.com

If someone can send me the files I would be very greatful!

jvictors-tp, 2023-11-27:
@aprime2t you normally need to compile it from source