osqzss/gps-sdr-sim #377: How to set PM time

HaXlezi, 2023-06-29:
Hello again, I want to simulate scenario start time which takes place in afternoon/night (PM time). However if I enter this time after _-t_ attribute in 24-hour format (e.g. 16:00), I got error message that max. value is 12:00. Is there some way, how to specify, that we want AM or PM time?
Thank you!

osqzss, 2023-06-29:
Use -T option.

HaXlezi, 2023-06-30:
Great, thanks!!

aprime2t, 2023-07-04:
Hello, I've been trying to download the gps-sdr-sim.zip file but it's missing from the release link. Could someone please send me the file? I've tried everywhere to try to find the file but with no luck!

My email is [aprime2t@gmail.com](mailto:aprime2t@gmail.com)

If someone can send me the files I would be very greatful! Or if someone could point me to where the files are to download that would be helpful too!

closed 2023-07-07 by HaXlezi